intel/zephyr.js #1399: [ocf] OcfClient can't connect to OcfServer

qiaojingx, 2017-07-25:
### Description

When running ocf server on A101, ocf client on Linux, the connection stuck. 
 
### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);
[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js);
[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js);
[OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js);
[OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js);
[OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js); 
[OcfDeviceInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfDeviceInfo.js); 
[OcfPlatformInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfPlatformInfo.js); 

### Steps to Reproduction

1. Build OcfServer on Arduino101.
2. Running OcfClient on Linux.

### Actual Result

 the connection stuck:
[image]

### Expected Result

OcfClient connect to OcfServer successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5726f4e | Arduino 101 | Jul 25, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-07-25:
I can't reproduce this, or at least not consistently, I thought I saw it happened once, but then I did a clean build and reflashed, and now I can't reproduce.

wanghongjuan, 2017-07-26:
@qiaojingx  I get the same result as @jimmy-huang with commit 1143544, the OcfClient.js(https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js) and [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js) can find the registered resource `/a/light/` in [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js), but the issue #1316 is reproduced when running `OcfClientUpdate.js`

qiaojingx, 2017-07-26:
It seems I was running into a bad situation. Verified with commit 1143544, this issue is not reproducible.
But other two issues are found. One is `Segmentation fault` error in [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js) tracking by #1407. Another one is ocfclient can't connect and disconnect with ocfserver repeatedly, tracking by #1408.
Close this one.

closed 2017-07-26 by qiaojingx