intel/zephyr.js #1400: [net-config] Function NetConfig.setBleAddress（） does not work

cuiyanx, 2017-07-25:
### Description

I set ble address as "F1:E2:D3:C4:B5:A6" in my code, but I get "FF:EE:DD:CC:BB:AA" when I search this ble device.
 

### Test Code

Simple test on a101:
```
var WebSocket = require("ws");
var netConfig = require("net-config");

var bleAddr = "F1:E2:D3:C4:B5:A6";
netConfig.setBleAddress(bleAddr);

console.log("Please connect a101 with Bluetooth....");
console.log("setBleAddress(bleAddr): expected result 'F1:E2:D3:C4:B5:A6'");
```

 

### Steps to Reproduction

1. Run simple test on a101
2. Connect a101 with bluetooth
3. Check console log
 

### Actual Result

Search a101 log:
```
root@intelProject:/home/cuiyan/intelwork/cy/zephyr.js# hcitool lescanLE Scan ...
43:47:8D:78:C8:4E (unknown)
43:47:8D:78:C8:4E (unknown)
FF:EE:DD:CC:BB:AA (unknown)
FF:EE:DD:CC:BB:AA Zephyr
57:5A:37:52:81:12 (unknown)
```
 

### Expected Result

Can set ble address.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5726f4ecaa | Arduino 101 | July 25, 2017 | Fail

 

### Additional Information

jimmy-huang, 2017-08-11:
The problem I think is that before setBleAddress is called, bluetooth is already enabled() and I don't think Zephyr allows up to update the MAC address once the stack is up, I think we need some kind of mechanism to grep if setBleAddress() is called in JS, then it should not call bt_enable() in the init, and wait till setBleAddress to called.  Or alternatively, expose a BLE.enable() method in JS that requires the app to call it once it sets the ble address.

jprestwo, 2017-08-11:
@jimmy-huang you could also set the address before you required any of the net modules, but thats kinda messy. The setBleAddress API should still work with OCF though, because bt_enable isn't called till later.

grgustaf, 2017-08-15:
@jimmy-huang I was you expecting to look at this one, not sure what you thought. ;) Anyway if I get to the end of my others I can look at it anyway.

Our idea was to add a "post-jerry-run" form of module init() function and move the bt_enable into there for both net_config at least, so that the BLE address function works. Bonus points for documenting that the function must be called in the initial run, not async/timeout. Also could throw an error if it's called late to help the user.

Later in the day in a discussion with @jprestwo he reminded me that you talk about "virtual BLE address" being used in BLE module - so this doesn't work there... but to simplify things we should probably make this work there too, if it's possible to set this "virtual" address. That doesn't have to be for 0.4, but if it's easy, maybe.

qiaojingx, 2017-08-18:
Verified with commit 4c98a2f3. This issue is fixed.
[image]

closed 2017-08-25 by grgustaf

cuiyanx, 2018-01-10:
Fixed and closed

PR #1484 (2017-08-25): [ble] Fixed not able to set BLE address
The bug is that once bt_enable() is called, it's impossible to set
the BLE address, so we have to move bt_enable() to be called after
all modules and the JS is loaded, this guarantees that any API call to
setBleAddress will be called before bt_enable().

Also update documentation to reflect this change.

Fixes #1400
commit bf70e366 (2017-08-25): [ble] Fixed not able to set BLE address (#1484)

The bug is that once bt_enable() is called, it's impossible to set
the BLE address, so we have to move bt_enable() to be called after
all modules and the JS is loaded, this guarantees that any API call to
setBleAddress will be called before bt_enable().

Also update documentation to reflect this change.

Fixes #1400