intel/zephyr.js #1401: [net-config] Function netConfig.setStaticIP() is undefined

cuiyanx, 2017-07-25:
### Description

I test net-config APIs and get an error. The function netConfig.setStaticIP() is undefined.
 

### Test Code

Simple test:
```
console.log("simple test");

var WebSocket = require("ws");
var netConfig = require("net-config");

var IPv6 = "2001:db8::100";
console.log("----mark----");
var IPState = netConfig.setStaticIP(IPv6);
console.log("setStaticIP(IPv6): " + IPState + " expected result 'true'");
```

 

### Steps to Reproduction

1. Run simple test on a101
2. Check console log.
 

### Actual Result

Log:
```
simple test
----mark----
Error: Expected a function.
```

 

### Expected Result

Work well.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5726f4e | Arduino 101 | July 25, 2017 | Fail

 

### Additional Information

jprestwo, 2017-07-25:
The 'p' in 'IP'  should be lowercase:

`netConfig.setStaticIp(IPv6);`

cuiyanx, 2017-07-26:
@jprestwo Well, I think we will update net-config.md

closed 2017-08-01 by grgustaf

cuiyanx, 2017-08-02:
fixed and closed

PR #1418 (2017-08-01): [net-config] Update setStaticIP() function API
Fixed #1401