intel/zephyr.js #1403: [net-config] Please optimize all API docs

cuiyanx, 2017-07-25:
### Description

In aio:
```
AIOPin.on
void on(string eventType, ReadCallback callback);
```
In ble:
```
BLE.on
void on(string eventType, EventCallback callback);
```
In gpio:
```
GPIOPin.onchange
attribute ChangeCallback onchange;
```
In net-config:
```
NetConfig.onup Event
void onup(void);

NetConfig.ondown Event
void ondown(void);
```
What is rule for event type? If I only read api docs, I will can not use it. 
So, please optimize all API docs. 

### Test Code

 

### Steps to Reproduction

 

### Actual Result

 

### Expected Result

 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 5726f4e | Arduino 101 | July 25, 2017 | Fail

 

### Additional Information

grgustaf, 2017-07-25:
Are you pointing out the inconsistency between using on() in one API but onchange in another? Yes, we should make them all use a common method.

Or are you really asking for something to change in the documentation alone? If so, I'm confused about what you're asking for.

grgustaf, 2017-07-25:
OK, @jprestwo helped me understand that you mean events are documented in different ways. I think we would probably be wise to move our documentation to be formatted more like Node.js documentation instead of WebIDL-like. So that has a specific way of handling Event documentation. Mission accepted. :)

grgustaf, 2017-08-15:
Fixed in #1470.

closed 2017-08-16 by jimmy-huang

cuiyanx, 2017-08-24:
Fixed and closed

PR #1470 (2017-08-16): [docs] Move to a common Node.js-like format for documenting events
Fixes #1403.
commit 0503562b (2017-08-16): [docs] Move to a common Node.js-like format for documenting events

Fixes #1403