intel/zephyr.js #1407: [ocf] Segmentation fault error occurs

qiaojingx, 2017-07-26:
### Description

When running [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js) on Linux, `Segmentation fault (core dumped)` error occurs.

### Test Code

[OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js)
[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);
 
### Steps to Reproduction

1. Build [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on Arduino101 via BLE.
2. Build [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js) on Linux

### Actual Result

`Segmentation fault (core dumped)` error occurs:
[image]

### Expected Result

No error occurs.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1143544 | Linux | Jul 26, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-07-31:
I can't reproduce this with master e6d1bf82aa4eb7bb12444d5a5f9a4a115ac63c7a

qiaojingx, 2017-08-01:
Closed since not reproduced with commit e6d1bf8.

closed 2017-08-01 by qiaojingx