intel/zephyr.js #1408: [ocf] ocfclient cannot connect to ocfserver after connecting and disconnecting 4-5 times

qiaojingx, 2017-07-26:
### Description

Try to connect and disconnect ocfclient with ocfserver, after repeating 4-5 times, ocfclient will be no longer able to connect to ocfserver.
 
### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);
[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js);
[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js);
[OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js);
[OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js);
[OcfSensorClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorClient.js); 
[OcfDeviceInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfDeviceInfo.js); 
[OcfPlatformInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfPlatformInfo.js); 

### Steps to Reproduction

1. Build OcfServer on Arduino101.
2. Running OcfClient on Linux.
3. CTRL+C to disconnect OcfClient.
4. Repeat step 2 and 3 for 4-5 times(OcfClient can be different client samples).

### Actual Result

After 4-5 times, ocfclient will be no longer able to connect to ocfserver, it stuck at `Started OCF client`:
[image]

### Expected Result

OcfClient can always connect to OcfServer successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 1143544 | Arduino 101 | Jul 26, 2017 | Fail
 
### Additional Information

Reset OcfServer can fix this issue, but it will be reproduced again after connecting 4-5 times.

wanghongjuan, 2017-07-31:
I just tested this with commit e40cd69, after running 10 times `OcfClientUpdate.js`, the client can connect the server, and the registered resource can be found/updated.

qiaojingx, 2017-08-01:
Closed since not reproduced with commit e40cd69.

closed 2017-08-01 by qiaojingx