intel/zephyr.js #1413: [net] net context bind error

qiaojingx, 2017-07-27:
### Description

This is a regression issue. Test with sample ([TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)) and test([test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js)), I get `net_context_bind` error even after adding router and starting [test-socket-server.py](https://github.com/01org/zephyr.js/blob/master/tests/tools/test-socket-server.py).
### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js)
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js)
 
### Steps to Reproduction

1. Build [test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js) on A101.
2. Connect A101 with PC via BLE.
3. Add router and IPv6 address on Linux:
      `ip -6 route add 2001:db8::/64 dev bt0`
      `ip -6 addr add 2001:db8::2/64 dev bt0`
4. Start [test-socket-server.py](https://github.com/01org/zephyr.js/blob/master/tests/tools/test-socket-server.py)
      
### Actual Result

 `net_context_bind` error:
[image]

### Expected Result

No error. TCPClient can connect with TCPserver. 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | Comment
---|---|---|---|---|---
master | 1143544 | Arduino 101 | Jul 27, 2017 | Fail |
master | 5c65f11 | Arduino 101 | Jul 27, 2017 | Pass | #1371, but no error if remove net.on('up'){}
master | 1143544 | FRDM-K64F | Jul 27, 2017 | Fail |
master | 5c65f11 | FRDM-K64F | Jul 27, 2017 | Pass|
### Additional Information

grgustaf, 2017-08-09:
Fixed in #1448

qiaojingx, 2017-08-10:
Verified with commit ddc30dc + #1448 on Arduino101, FRDM-K64F and qemu. This issue is not reproduced.

PR #1448 (2017-08-10): [net-config] Update docs to require setting IP address
If the net-config module is included, the user needs to either set a
static IP address or use DHCP; no default address will be set. We could
potentially change this, but this requirement makes for better example
code because it's more realistic.

Added setStaticIP call to a sample and a test that needed it.

Fixes #1413
commit f0020301 (2017-08-10): [net-config] Update docs to require setting IP address (#1448)

If the net-config module is included, the user needs to either set a
static IP address or use DHCP; no default address will be set. We could
potentially change this, but this requirement makes for better example
code because it's more realistic.

Added setStaticIP call to a sample and a test that needed it.

Fixes #1413
closed 2017-08-10 by jimmy-huang