intel/zephyr.js #1415: [ocf] Fatal fault occurred when running sample with Grove Chainable RGB LED sensor

wanghongjuan, 2017-07-28:
## Description
I an writing a sample using OCF API with [Grove Chainable RGB LED](http://www.seeedstudio.com/depot/twig-chainable-rgb-led-p-850.html?cPath=156_157) sensor, there is a fatal fault occurred when update the state of the sensor at second time.

## Test Code
**Server**: [rgbled.js](https://github.com/01org/zephyr.js/files/1182678/rgbled.txt)
**Client**: [client.js](https://github.com/01org/zephyr.js/files/1182681/rgbled-client.txt)

## Steps to Reproduce
1. Build rgbled.js on A101.
2. Connect with BLE
3. Build and run client.js on linux, after getting the output, exit the process

## Actual Result
This is the log of master:

[image]

I also try to run this sample with commit 8d1b343, get more detail information as below:

[image]

The client is ok:
[image]

## Excepted Result
There is no error no matter how many times that update the RGB LED resource's state

grgustaf, 2017-07-28:
@jimmy-huang feel free to lower priority but first take a look at it.

wanghongjuan, 2017-07-31:
@jimmy-huang  I still can reproduce this issue. The led seems doesn't light, I updated the client code that only invoke the `updateResource(data['/a/rgbled'], [255,255,255])` one time, all code runs normally, no error, and I checked the led again, the led is fine.

jimmy-huang, 2017-07-31:
@wanghongjuan, you getting a segfault in client or the server?  For the client, you might wanna try pr #1416 if it helps.

wanghongjuan, 2017-08-02:
@jimmy-huang I tried the pr and got undefined when invoking the function zjs_create_promise directly in client.js, am i missing something?

jimmy-huang, 2017-08-03:
@wanghongjuan the #1416 may not be ready, but try #1424

wanghongjuan, 2017-08-03:
@jimmy-huang #1424 is work, thanks for you support.

closed 2017-08-03 by wanghongjuan