intel/zephyr.js #1422: [ws] Parameters in websocketconnection.onclose are undefined

qiaojingx, 2017-08-02:
### Description

Code and reason in onclose event are as undefined, when websocketconnection.onclose is triggered. Besides, description of onclose event in [Web IDL](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#web-idl) and [API Documentation](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#api-documentation) are not consistent. One has parameters and the other one doesn't.
[Web IDL](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#web-idl):
```
interface WebSocketConnection {
    // WebSocketConnection Events
    onmessage(Buffer data);
    onping(Buffer data);
    onpong(Buffer data);
    onclose(Number code, String reason);

```
[API Documentation](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#api-documentation):
```
WebSocketConnection.onclose Event

onclose(void)

The close event is emitted when the web socket has closed.
```

### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js);
[test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js);
 
### Steps to Reproduction

1. `make JS=tests/test-ws-server.js ROM=256`
2. `make dfu`
3.  connect via ble and add soft router on linux:
    ` ip -6 route add 2001:db8::/64 dev bt0`
4. run ws client on linux by `node /tests/test-ws-client-node.js` 
5. CTRL+C to disconnect websocket connection.

### Actual Result

code and reason display as undefined.
[test-ws-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws-server.js):
[image]
[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js):
[image]

### Expected Result

1. Should these two parameters be defined as [Web IDL](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#web-idl) said? Or just remove this test point and update onclose event in [Web IDL](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#web-idl) as no parameters.

2. Description of onclose event in [Web IDL](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#web-idl) and [API Documentation](https://github.com/01org/zephyr.js/blob/master/docs/web-socket.md#api-documentation) should be consistent.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 4d9b084 | Arduino 101 | Aug 2, 2017 | Fail
 
### Additional Information

brianjjones, 2017-08-04:
This is fixed by https://github.com/01org/zephyr.js/pull/1423

closed 2017-08-08 by jimmy-huang

qiaojingx, 2017-08-09:
Verified with commit ca7af31 on Arduino101, FRDM-K64F and Linux. This issue is not reproduced.

PR #1423 (2017-08-08): [websockets] Fix close connection after event patch
Fixes #1422
commit 2481c6d4 (2017-08-08): [websockets] Fix close connection after event patch (#1423)

Fixes #1422