intel/zephyr.js #1428: OCF resources with multiple sensors not found

qiuzhong, 2017-08-03:
## Description:
When there is only a sensor, the OCF resource can be discovered.
Where there are multiple sensors, the OCF resources cannot be discovered.

## Steps to reproduced:
1. Build [SmartHome-Demo/ocf-servers/zjs-servers/multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js) sample.
2. Flash it to the 101 board and connect 6LowPAN by follow [this](https://github.com/01org/SmartHome-Demo/tree/master/ocf-servers/zjs-servers#establish-a-ble-connection-from-gateway)
3. Install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server), then run `$ cd iot-rest-api-server/test && ./oic-get /res`

## Actual Result:
OCF resource not found.

## Expected Result:
All OCF resources can be found.

## Test Builds
| Branch | Commit Id | Target Device  | Example | Sensor number | Result |
| --- | --- | --- | --- | --- | --- |
| master | [e40cd69](https://github.com/01org/zephyr.js/commit/e40cd69d18420ccc2c14c26633c8daa676ccaf66) | Arduino 101 | [led.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/led.js) | 1 | PASS | 
| master | [e40cd69](https://github.com/01org/zephyr.js/commit/e40cd69d18420ccc2c14c26633c8daa676ccaf66) | Arduino 101 | [multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js) | 6 | FAILED |

jimmy-huang, 2017-08-03:
There are issue with resource limitation now running with OCF server, I think it still works up till 3 resources, but anything over 3 won't work, it looks like we are running out of memory in the bluetooth stack when sending out packets.

ttzeng, 2017-08-04:
**Input**: I tried to mask off all resource registrations except one in the multiserver test, logically only one resource is registered and only its property should be responded to the client, but even this single (logically) resource can not be enumerated by the REST API server (/res), not mentioning the access to its property.

jimmy-huang, 2017-08-04:
Here's  fix the get around the limitation see #1430, you'll need to increase the CONFIG_NET_BUF_DATA_SIZE to 512, and then increase the RAM, I am not sure if we will merge this patch, since I think it's the app that should tweak those settings.

So when you build ZJS, you can pass in 

make JS=multiserver.js RAM=64

gvancuts, 2017-08-07:
Hi @jimmy-huang , thanks for the pointer to the fix!

Can these really be tweaked from the application (short of rebuilding ZJS completely)?

jimmy-huang, 2017-08-07:
@gvancuts, not really, you'll have to rebuild the app, since they are statically defined in the prj.conf, there's no way to dynamically change them at runtime.  And these might just be settings to mediate some of the problems, there could be others that might have to be tweaked for BLE, as 6lowpan seems to be the problem here.  I filed a bug on iotivity-contrained https://github.com/iotivity/iotivity-constrained/issues/10 also, so maybe someone from that team might have better insights on what the optimal settings are if there's any.  I am guessing you'll just have to play around with these settings:

CONFIG_NET_PKT_RX_COUNT=5
CONFIG_NET_PKT_TX_COUNT=5
CONFIG_NET_BUF_RX_COUNT=10
CONFIG_NET_BUF_TX_COUNT=10
CONFIG_NET_BUF_DATA_SIZE=512

closed 2017-08-07 by grgustaf

qiuzhong, 2017-08-10:
Verified with [ddc30dc](https://github.com/01org/zephyr.js/commit/ddc30dc9db3963977eac933be148ffc54c23aec2), this bug is not reproduced again.

wanghongjuan, 2017-08-10:
Verified this issue with commit ddc30dc , it cannot be reproduced.

PR #1430 (2017-08-07): [ocf] Increase net data and ram to handle multiple OCF resources
The Smart Home demo registers up to 6 resources, which the current
network data setting is not capable of handling over BLE, this
patch increase the CONFIG_NET_BUF_DATA_SIZE value to 512 and
also more RAM.

Fixes #1428
commit 1d78d245 (2017-08-07): [ocf] Increase net data and ram to handle multiple OCF resources (#1430)

The Smart Home demo registers up to 6 resources, which the current
network data setting is not capable of handling over BLE, this
patch increases the CONFIG_NET_BUF_DATA_SIZE value to 512 and
also more RAM.

Fixes #1428