intel/zephyr.js #1432: [event] Event.js failed after rework event APIs

qiaojingx, 2017-08-07:
### Description

 Running [Event.js](https://github.com/01org/zephyr.js/tree/master/samples/tests/Event.js) failed after rework event APIs.  `test_event1` is treated as `test_event` in [Event.js](https://github.com/01org/zephyr.js/tree/master/samples/tests/Event.js).

### Test Code

[Event.js](https://github.com/01org/zephyr.js/tree/master/samples/tests/Event.js);
 
### Steps to Reproduction

 1. `make JS=samples/tests/Event.js`
 2. `make dfu`
### Actual Result

 
[image]

### Expected Result

 sample test pass.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | f3625a8 | Arduino 101 | Aug 7, 2017 | Fail
master | f3625a8 | FRDM-K64F | Aug 7, 2017 | Fail
master | f3625a8 | Linux | Aug 7, 2017 | Fail
master | e40cd69 | Arduino 101 | Aug 1, 2017 | Pass
master | e40cd69 | FRDM-K64F | Aug 1, 2017 | Pass
master | e40cd69 | Linux | Aug 1, 2017 | Pass
### Additional Information

cuiyanx, 2017-08-07:
I rename the event `test_event1` to `event_test` and remove` -1 `on 34 line, this sample work well.
```
EventName[0]: test_event
EventName[1]: event_test
testListenerNames.length = 2
testListenerCount.length = 2
test_event: handler 1: arg1=1234 arg2=4567
test_event: handler 2: arg1=1234 arg2=4567
event_test: arg1=1000 arg2=1111
test_event: handler 2: arg1=1234 arg2=4567
Listeners for test_event = 1
Listeners for test_event after removeAllListeners = 0
Event test passed
```

grgustaf, 2017-08-07:
Thanks, I provided a simpler fix in #1435 though.

closed 2017-08-07 by jimmy-huang

qiaojingx, 2017-08-09:
Verified with commit ca7af31 on Arduino101, FRDM-K64F and Linux. This issue is not reproduced.

PR #1435 (2017-08-07): [events] Stop matching on event prefix
This compares the null terminator of the event name string so that it
will no longer match prefix but only exact string match.

Thanks to Cui Yan for finding the bug and another proposed solution.

Fixes #1432
commit 9a272ee2 (2017-08-07): [events] Stop matching on event prefix (#1435)

This compares the null terminator of the event name string so that it
will no longer match prefix but only exact string match.

Thanks to Cui Yan for finding the bug and another proposed solution.

Fixes #1432