intel/zephyr.js #1445: [spi] Read and write 8 bits data failed if set 16 bits for SPI

qiaojingx, 2017-08-09:
### Description

When running [test-spi.js](https://github.com/01org/zephyr.js/blob/master/tests/test-spi.js), read and write 8 bits data failed if set 16 bits for SPI.
 
### Test Code

[test-spi.js](https://github.com/01org/zephyr.js/blob/master/tests/test-spi.js);
 
### Steps to Reproduction

 1. `make JS=tests/test-spi.js`
 2. `make dfu`
### Actual Result

 [image]

### Expected Result

 No error 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ca7af311 | Arduino 101 | Aug 9 2017 | Fail
master | ca7af311 | FRDM-K64F | Aug 9 2017 | Fail 
### Additional Information

jimmy-huang, 2017-08-09:
Since @brianjjones is out, I did some research on this, It looks like this can be fixed if you set the busSpeed back to 20, in Mhz, which it was originally set, but this doesn't work on the FRDM board as it would hit an error

assertion "false" failed: file zephyr.js/deps/zephyr/ext/hal/nxp/mcux/drivers/fsl_dspi.c", line 188, function: DSPI_MasterInit

I believe this is why it was later tweaked to 20000 by a patch from @cuiyanx, but I think we should set it to 20 instead, maybe setting it to 20000 is not the right fix to get it working on the FRDM board. do you know how did you pick 20000 to be the speedBus?  Maybe @jprestwo or @grgustaf  would have some insights on this.

jimmy-huang, 2017-08-10:
@cuiyanx in the test, shouldn't the 2nd element in the last test be "010203", not "010200"?  at least it fixes on frdm-k64f, still failng on Arduino 101 though, not sure why.

jimmy-huang, 2017-08-11:
With patch #1452 and #1449, it will pass FRDM but still fails Arduino 101, the spi_transceive() call in Zephyr is indeed returning the wrong values randomly depending on timing, and I think we can also try updating to use the new SPI api, as this might be a bug of the current legacy API.

jimmy-huang, 2017-08-11:
Patch #1452 and #1449 address some of the issues related to this bug, but I've looked into it a little bit and it's the zephyr native spi_transceive() api that's returning the invalid values, and the receiving value seemed to be random,  Maybe on the Arduino 101, only 8bit word size is supported, 16bit is not, @brianjjones can look into this when he's back.

jimmy-huang, 2017-08-11:
One more note, I am not sure if SPI can really support 16bit words, since most of the SPI samples uses 8bit, and given that the tests actually sends/reads 3 bytes "010203", which is not in 16bit chunks, the behavior is unknown, maybe the test case is invalid? @jprestwo do you know?

brianjjones, 2017-08-25:
I think the test that is failing is invalid.  The array contains 24 bits, there's no way to send that in 16 bit chunks without padding it with 0's somewhere.  And doing that could easily change the meaning of the data being sent.  I could add a more useful error for when the user tries to transceive less than the full number of bits. @grgustaf  thoughts?

brianjjones, 2017-08-25:
Also note the part that fails is the Zephyr call spi_transceive, so if we think its a valid use case we'll need to get it changed in Zephyr.

PR #1476 (2017-08-25): [spi] Remove test point for test-spi.js
Fixes #1445
closed 2017-08-25 by grgustaf