intel/zephyr.js #1450: [sensor] Gyroscope and temperature sensors are not supported on K64F

qiaojingx, 2017-08-10:
### Description

W3C Sensors should support on-board sensors on the FRDM-K64F according to feature #1320. But gyroscope and temperature sensors are still not supported on K64F. They are undefined.

### Test Code

[BMI160Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Gyroscope.js);
[BMI160Temperature.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Temperature.js);

### Steps to Reproduction

1. `make BOARD=frdm_k64f JS=samples/BMI160Gyroscope.js`
2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`

### Actual Result

 [image]

[image]

### Expected Result

Gyroscope and temperature sensors are supported on K64F
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ddc30dc | FRDM-K64F | Aug 10, 2017 | Fail
 
### Additional Information

jimmy-huang, 2017-08-10:
Yeah, we don't currently support any on-board temperature or magnetometer sensors with w3c sensors for the frdm board.

haoxli, 2017-08-11:
Thanks for your clarification.  @qiaojingx please close this one.

closed 2017-08-11 by qiaojingx