intel/zephyr.js #1451: [sensor] incorrect update frequency / sampling rate of the Sensor API implementation

ttzeng, 2017-08-10:
The [IDL of the ZJS Sensor interface](https://github.com/01org/zephyr.js/blob/master/docs/sensors.md#web-idl) states the sampling rate 'frequency' is in Hz, however if we changed the sampling rate of the [Accelerometer](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Accelerometer.js), [Gyroscope](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Gyroscope.js), or [Temperature](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Temperature.js) sample to 1 Hz, the actual update frequency is faster than every 1 second.

grgustaf, 2017-08-10:
@jimmy-huang please look into what the limits of the frequency are. If very low frequency isn't possible, maybe we could implement it with timers.

jimmy-huang, 2017-08-10:
@ttzeng can you clarify which sample you were running, was this in the ashell/ide or just build and flash on the device?  Because I think the ashell/ide use the polling while the flash image uses triggers.

ttzeng, 2017-08-12:
@jimmy-huang thanks for looking into this. Hereby I confirm this issue has been fixed by #1462 :+1:

PR #1462 (2017-08-15): [sensor] Fixed polling frequency not updated
The poll frequency was declared as a static variable and
couldn't be set from a different src file, making it extern
so it could be set.

Fixes #1451
commit 95f6ff4b (2017-08-15): [sensor] Fixed polling frequency not updated (#1462)

The poll frequency was declared as a static variable and
couldn't be set from a different src file, making it extern
so it could be set.

Fixes #1451
closed 2017-08-15 by grgustaf