intel/zephyr.js #1453: Update SPI module to use new Zephyr spi API

jimmy-huang, 2017-08-10:
The current zephyr SPI api used in the SPI module is the legacy api, and we should switch to the new SPI api, as the new API address some of the issues like ZEP-1725, where the frequency invalidly set max_sys_feq as a clock divider number like on the Arduino 101.

brianjjones, 2017-08-22:
This is because currently the Arduino 101 SPI driver uses the legacy SPI, not the new one.  I already ported this code to use the new API, but it won't work until the driver gets updated.  But when that happens, should be a quick easy update.

grgustaf, 2018-02-05:
@brianjjones, @jimmy-huang what's the status of this?

brianjjones, 2018-02-05:
@grgustaf The SPI driver for the A101 and FRDM_K64F are both still using the Legacy API, possibly the same for other boards.  So until these drivers are updated, there's no way for us to update the SPI API without breaking the build for several of our key boards.  Perhaps we should ping someone on the Zephyr team about driver plans.