intel/zephyr.js #1454: [sample] Error name and host address are not applicable in TCPClient

qiaojingx, 2017-08-11:
### Description

Error name in TCPClient is not `NotFoundError ` anymore, it is just `Error` now.
Should the host address in [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) be `2001:db8::2` since TCP python server can't be created by address like `fe80::5ef3:70ff:fe78:1080` ?

### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js); 

### Steps to Reproduction

1. insert `console.log(err.name, ":", err.message);` to line 48 in [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
2. Build [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) on Arduino101.
3. Connect PC via BLE and add  a new route for the bt0 interface:
` ip -6 route add 2001:db8::/64 dev bt0`
4. wait for error occur

### Actual Result

1.  [image]

2. If host name  keeps as `fe80::5ef3:70ff:fe78:1080`. TCP connection won't setup.

### Expected Result

Error name match the actual situation.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e3e4dde | Arduino 101 | Aug 11, 2017 | Fail
 
### Additional Information

grgustaf, 2017-08-11:
We probably need to look at this in more detail, but the number of distinct "error types" being used seemed unnecessary so I did make changes here. Off-hand, I think the only reason to give distinct error types is if we really expect the user to check for these types and react differently in their code. One way to accomplish that is to start with everything as Error and then make errors more specific once we prove there is an important use case for the distinction. When we do so we should describe the reason in a code comment so it doesn't just get changed again in the future on a whim.

Otherwise, the error _message_ can contain relevant detail to let the user know what's going wrong. Usually this helps them find a coding error and once they find it they don't really need to special case that error. So I think I was reacting to the silliness of something like "NotFoundError: host not found" which repeats itself.

qiaojingx, 2017-09-15:
Verified with commit 174f35f . This sample has been updated.
Close.

closed 2017-09-15 by qiaojingx