intel/zephyr.js #1455: [tcp] tcpclient can't connect to tcpserver if server start later than client

qiaojingx, 2017-08-11:
### Description

 tcpclient can't connect to tcpserver if server start later than client over 10s.

### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);

tcpserver.py
```
# /usr/bin/python
# -*- coding: utf-8 -*-
import socket
import time
sock = socket.socket(socket.AF_INET6, socket.SOCK_STREAM)
server_address = ('2001:db8::2', 4242)
print "Starting up on %s:%s" % server_address
sock.bind(server_address)
sock.listen(1)
print "Waiting for a connection"
connection, client_address = sock.accept()
print "Connection from", client_address
while True:
        data = connection.recv(1024)
        print "Receive '%s'" % data
        connection.send(data)
        time.sleep(3)
```
### Steps to Reproduction

1. Change host address in line 52 and 59 to `2001:db8::2` and change err.name in line 48 to `err.name == "Error"` in [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) because of #1454 ;
2. Build [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) on Arduino101.
3. Connect PC via BLE and add  a new route for the bt0 interface:
` ip -6 route add 2001:db8::/64 dev bt0`
`ip -6 addr add 2001:db8::2 dev bt0`
4. After 10s, start tcpserver by `python tcpserver.py`

### Actual Result

tcpclient can't connect to tcpserver
[image]

### Expected Result

tcpclient could connect to tcpserver whenever server started.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e3e4dde | Arduino 101 | Aug 11, 2017 | Fail
 
### Additional Information

grgustaf, 2017-08-11:
I haven't looked at this yet, but clients expects servers to exist already. So this sounds like reasonable behavior, but we could also change the client to retry forever if we want. I assume it's just trying a certain number of times and then giving up.

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b in master. This issue is not reproduced on Arduino101 over Ethernet, FRDM-K64F and Qemu.
But on Arduino101 over ble, this issue cannot be verified because of issue #1518.