intel/zephyr.js #1457: [ocf] Sometimes Ocf server resources cannot be found when running multiple sensors

wanghongjuan, 2017-08-11:
## Description
Sometimes Ocf server resources cannot be found when running multiple sensors

## Steps to Reproduce
1. Build sample [multiple.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js) and flash it to Arduino 101.
2. Connect to BLE, install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server/blob/master/index.js)
3. Run `$ cd iot-rest-api-server/test && ./oic-get /res`, after getting all ocf resources, pick one of them, run `./oic-get /a/pir?di=[xxxxxx]` 6~8 times

## Actual Result
The iot-rest-api-server hangs, wait a moment, show `Resource retrieve failed`. Run `./oic-get /res` again, all registered resources are gone, but the 6lowpan is still connection.

## Expected Result
The registered resources always should  be found

## Test Build
| Branch | Commit ID | Target Device | Test Date | Result |
|--------|--------|--------|--------|---------|
| master | e3e4dde | Arduino 101 | Aug 11, 2017 | Failed |

jimmy-huang, 2017-08-15:
I can't even get any resources using ./oic-get /a/pir?di=[xxxxxx], it will always return an error
Resource retrieve failed: get is not defined, I am wondering if there's a iot-rest-api-server bug

wanghongjuan, 2017-08-17:
The iot-rest-api-server just released new version 0.6.0 which based on iotivity-node v1.3.0, this version has a lot of changes. In this issue, I am using iot-rest-api-server@0.5.0, which version are you using? The motion resource has been registered in [mutilple.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js#L237), it should can be found via `./test/oic-get 
 /res`.

jimmy-huang, 2017-08-17:
Yeah, @wanghongjuan the problem is that iotivity v1.3.0 has enabled secure mode by default, and none of our OCF samples are build with secure mode, because they won't fit on the Arduino 101.  @nagineni has looked into this a little bit, I think there are issues with discovery when you use iot-rest-api-server with has secure mode and our ZJS ocf server, which does not.

wanghongjuan, 2017-08-21:
@jimmy-huang I think your explanation more suit #1481 , but this issue is not relate to iotivity v1.3.0. Could you reproduce this issue with iot-rest-api-server@v0.5.0 at present?

jimmy-huang, 2017-08-23:
@wanghongjuan I think this issue is related to iot-rest-api-server, can you try the following patch by @nagineni to see if it fixes your issue?  At least it works for me.

https://github.com/nagineni/iot-rest-api-server/commit/6666405e76b5866b0346348ce87caa4034410a8c

nagineni, 2017-08-24:
I found an issue in iotivity-node that we are doing full resource discovery on retrieve request and that might be the reason for this failure. There might not be enough free TX/RX buffers on device side to accommodate the response of the resource discovery on retrieve. 

We are working on removing the unneeded auto resource discovery on retrieve now. Hope this issue will disappear with that fix.

wanghongjuan, 2017-08-25:
Hi, @jimmy-huang @nagineni  I tried the commit you mentioned with iot-rest-api-server@0.5.0, when run `./oic-get /a/temperature?di=[xxxxxx]` 6~8 times, reproduce this issue again.

jimmy-huang, 2018-01-11:
@wanghongjuan is this still reproducible with latest Zephyr 1.10 and iotvity-constrained?

xiaoyu3x, 2018-01-12:
@jimmy-huang 
This issue can't be verified now because issue #1664 hasn't been fixed.
We can't build  multiple sensors image successfully now.
I will verify it until issue #1664 is resloved.

jimmy-huang, 2018-01-23:
@xiaoyu3x Is this reproducible with K64F?

jimmy-huang, 2018-01-23:
@xiaoyu3x Can you retest this with #1785, and using make JS=multiserver2.js ROM=260?  multiserver2.js doesn't require ARC image, so you can bump up the size even further to get around the size problem.

xiaoyu3x, 2018-01-24:
hi, @jimmy-huang 
This issue isn't reproduced when using multisever.js on K64F.
This issue isn't reproduced when using `make ROM=256 JS=multiserver2.js` with #1785  on A101.

jimmy-huang, 2018-01-24:
@xiaoyu3x With patch #1789, you can further bump ROM_SIZE to 265, with multiserver.js I can set as high as 272, can you re-test?

xiaoyu3x, 2018-01-25:
hi, @jimmy-huang 
There are six sensors in multiserver.js and they are "Grove light sensor/temperature/motion/button/buzzer/mini fan".
There is only 'Grove light sensor' is found successfully.
[image]
Glad to see this issue is not reproduced and it can be found every time when using #1789 

Unfortunately, other sensor can't be found totally.
For example:
[image]
[image]

I think this appearance is belong to another issue and I will file a new bug to trace.

jimmy-huang, 2018-01-25:
Yeah, maybe close this one and open another bug to track the other issue.

xiaoyu3x, 2018-01-25:
This issue isn't reproduced, so closed it.

jimmy-huang, 2018-01-25:
@wanghongjuan I am going to close this and please re-open if there are new issues.

closed 2018-01-25 by jimmy-huang