intel/zephyr.js #1458: [ws] BUS FAULT error when running dhcp websocket

qiaojingx, 2017-08-11:
### Description

This is a regression issue. When running websocket-dhcp, got `BUS FAULT` error on FRDM-K64F.

### Test Code

[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js);
[test-ws4-server-dhcp.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws4-server-dhcp.js);
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js);

### Steps to Reproduction

1. Build [WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js) on FRDM-K64F.
3. Connect to a router which assigns a DHCP address
4. `node node samples/websockets/NodeWebSocketClient4.js <ip address> <port>`

### Actual Result

BUS FAULT error:
[image]

### Expected Result

No error
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e3e4dde | FRDM-K64F | Aug 11, 2017 | Fail
master | f3625a8 | FRDM-K64F | Aug 11, 2017 | Pass

### Additional Information

grgustaf, 2017-08-11:
@qiaojingx, just wanted to comment that you guys do a really great job and in particular giving us a commit id where it passed and then failed for a regression is really helpful. :)

qiaojingx, 2017-08-14:
I found BUS FAULT error begins with commit 46d73b5. A little different is:  `(ERROR) no handle found
` will print when first connect, then bus fault error occurs when second connect.
[image]
Hope it is also helpful. :)

grgustaf, 2017-08-29:
The no handle found issue should be fixed by #1503; please retest.

grgustaf, 2017-08-29:
Oops that was net, this is ws - I'll need to check this out. It might well be the same type of problem that I just fixed in net.

brianjjones, 2017-08-30:
I tested it and see the same thing qiaojingx did.  I didn't get a chance to start trouble shooting it, but its easily reproducible.

qiaojingx, 2017-09-01:
Verified with a0b1217 + #1513. It only fixes `bus fault` issue in [WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js); But, other two tests([test-ws4-server-dhcp.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws4-server-dhcp.js); [test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js;)) still get `bus fault` issue when exit node by `ctrl+c`

Besides, `BUS FAULT` error is also caused in [UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js), [UDPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ4.js) and [test-ws6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws6-server.js)

grgustaf, 2017-09-01:
Fixed by updated #1513

closed 2017-09-05 by grgustaf

qiaojingx, 2017-09-11:
Verified with commit 5403f92. This issue is not reproduced.

commit 49f13c33 (2017-09-05): [callbacks] Move LOCK/UNLOCK to callbacks.c and use mutex instead

Not sure why, but this fixes a weird reboot issue I was seeing on
K64F while trying to fix #1458.
commit cf5951e2 (2017-09-05): [ws] Check for unconnected state when connection closes

If a payload was too large during the websocket handshake, the
connection would not be accepted so the connection JS object was
never created, but this wasn't being handled correctly later when
the connection closed.

Fixes #1458