intel/zephyr.js #1459: [ws] Error calling listener will be printed sometimes after closing websocket

qiaojingx, 2017-08-11:
### Description

This is a problem with a reproduction rate about 50%. After closing websocket by disconnect NodeWebsocketClient.js, `(ERROR) error calling listener` will be printed sometimes.

### Test Code

[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[NodeWebSocketClient.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js);
### Steps to Reproduction

1. Build [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) on Arduino101.
2. Connect PC via BLE and add  a new route for the bt0 interface:
` ip -6 route add 2001:db8::/64 dev bt0`
`ip -6 addr add 2001:db8::2 dev bt0`
3. `node samples/websockets/NodeWebSocketClient.js`
4. CTRL+C to disconnect NodeWebSocketClient.js

### Actual Result

[image]

### Expected Result

No error print.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e3e4dde | Arduino 101 | Aug 11, 2017 | Fail
 
### Additional Information

LWY931119, 2017-09-06:
When I do 'node samples/websockets/NodeWebSocketClient.js' the terminal says "Error: cannot find module 'net-config' "Do I miss any initial setup ?
By the way ,I did 'nc 2001:db8::1 8080' and it failed,Linux cannot Cannot establish TCP connection with arduino101.

qiaojingx, 2017-09-06:
@LWY931119 , it is strange because NodeWebSocketClient.js doesn't require ‘net-config’ module. Can you retest if it is 'ws' module?
If cannot find module 'ws', you need to install 'ws' module for node manually on Linux.
Websocket Server cannot be connected by `nc 2001:db8::1 8080`, it should be connected by `node samples/websockets/NodeWebSocketClient.js`
For TCP Server [TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js), it can be connected by `nc -6 2001:db8::1 4242`.

LWY931119, 2017-09-06:
Sorry  it is my fault. I executed WebSocketServer.js actually. Howerver, the NodeWebSocketClient.js failed too. 
It says error: ERROR connect ETIMEDOUT 2001:db8::1 8080 {"code":"ETIMEDOUT","errno":"ETIMEDOUT","syscall":"connect","address":"2001:db8::1","port":"8080"}.
By looking at wireshark , Arduino101 dose not  ACK the Linux's SYN

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b in master. This issue is not reproduced. Close.

closed 2018-01-25 by qiaojingx