intel/zephyr.js #1467: [PnP]Helloworld, I2C and  WebBluetoothGroveLcdDemo RAM samples increase obviously compared with master commit f3625a8.

hanyia, 2017-08-15:
Test with latest master commit on zephyr board, found RAM samples has more than ~5% gap compared with the last commit. SDK version is v0.9.1. Detail info as below table:

Device | KPI | ww33 master: e3e4dde | ww32 master: f3625a8 | ww33 vs ww32
-- | -- | -- | -- | --
Arduino101 | Zephyr_Helloworld_RAM(Byte) | 26752 | 24704 | 8.29%
Arduino101 | Zephyr_I2C_RAM(Byte) | 28800 | 26752 | 7.66%
Arduino101 | Zephyr_WebBluetoothGroveLcdDemo_RAM(Byte) | 38336 | 36288 | 5.64%

grgustaf, 2017-08-15:
This is due to commit 776330f, where we changed the stack size from 2K to 4K. It might be worthwhile to keep track somewhere of the fact that 4K comes from stack allocation, because sometime in the future when we're needing more RAM we should know that this one of the places to find it.

But basically this is expected and worth the tradeoff to improve BLE stability.

hanyia, 2017-08-22:
It is expected, so close this bug.

closed 2017-08-22 by hanyia