intel/zephyr.js #1468: [tcp] 'no handle found' error happens when running TCPEchoServ4_DHCP.js

qiaojingx, 2017-08-15:
### Description

This is a regression issue. When running [TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js), got `no handle found ` error on FRDM-K64F.

### Test Code

[TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js);

### Steps to Reproduction

1. Build [TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js) on FRDM-K64F.
3. Connect to a router which assigns a DHCP address
4. `echo "hello" | nc <DHCP ip> 4242`

### Actual Result

no handle found error occurs when try to connect with net-tools. if try to connect many times, it also can reproduce bus fault error.
[image]

### Expected Result

No error
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | e3e4dde- 46d73b5 | FRDM-K64F | Aug 15, 2017 | Fail
master | 9f75ded | FRDM-K64F | Aug 15, 2017 | Pass

### Additional Information

Not sure if it is the same as #1458 ,but they are caused by the same commit 46d73b5.

jimmy-huang, 2017-08-22:
I am also seeing this when I run the BLE.js sample on the nRF52 board, but not on the Arduino 101:

BLE sample...

src/zjs_event.c:397 zjs_emit_event():
(ERROR) no handle found

qiaojingx, 2017-08-23:
Verified with commit 06cce05 on FRDM_K64F. `Cannot send data to peer(-9)` is print instead of `No handle found` error when send data by net-tools. But `BUS FAULT` error still occurs after connect with net-tools for many times.
[image]
[image]

closed 2017-08-29 by grgustaf

qiaojingx, 2017-08-31:
Verified with commit a0b1217, this issue is not reproduced. But another issue is found, tracking by #1512

commit f08230ab (2017-08-29): [net] Acquire server until closed; store connections in server

The server object was being freed immediately if JS lost its reference
to it; instead it should only be freed if explicitly closed and all its
connections are closed.

To track this properly, store open connections within the server handle
they are associated with, and store a global list of servers instead of
connections.

Fixes #1468