intel/zephyr.js #1481: [ocf] ZJS OCF servers can't be discovered by OCF clients running 1.3.0 stack

ttzeng, 2017-08-19:
After iotivity-node push to [1.3.0-0](https://github.com/otcshare/iotivity-node/commit/f070902df26e233bc4d3214ca6c10a7d4b7ba8e1), the ZJS servers (e.g. samples/OcfServer.js) can't be discovered by the iotivity-node based Clients (e.g. iot-rest-api-server).

How to reproduce:
(1) build the ZJS image with the following command
`$ make BLE_ADDR='de:ad:be:ef:12:34' ROM=256 JS=samples/OcfServer.js`
(2) flash the image to A101 and reboot the A101
(3) install the `iot-rest-api-server`:
`$ npm install iot-rest-api-server`
(4) establish a 6lowpan connection to the A101
(5) discover OCF resources through iot-rest-api-server
(6) A101 should receive the request and send back the response, and display similar log in the serial console
```
oc_network_receive: received 26 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5e51:4fff:fe0e:c6cf]
oc_send_buffer: outgoing message: [fe80:0000:0000:0000:5e51:4fff:fe0e:c6cf]
```
However, after upgrading to iotivity-node 1.3.0-0, A101 receives requests but no responses
```
oc_network_receive: received 61 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5e51:4fff:fe0e:c6cf]
oc_network_receive: received 32 bytes
oc_network_receive: incoming message: [fe80:0000:0000:0000:5e51:4fff:fe0e:c6cf]
```

jimmy-huang, 2017-08-21:
@ttzeng Iotivity-node is enabling secure mode by default, but ZJS OCF does not support secure mode due to size limitation.  Also @nagineni  has pointed out that we will need to update iotivity-constrained to the latest 1.0 spec and he's working on a patch to resolve some of the discovery issues.   Eventually, we might have to figure out how to enable secure mode on Arduino 101, if that may not be possible with the limited ROM space and we might have to only support it on other boards like the FRDM-K64F.   For demo purposes, maybe you can try disabling secure mode on iotivity-node?

ttzeng, 2017-08-22:
@jimmy-huang I believe clients (iot-rest-api-server) with secured stack should be able to access unsecured servers (ZJS), as plain text coap:// is used and there is no SVR in the ZJS server that could filter out the requests. For example, there is a [simpleserver](https://github.com/iotivity/iotivity-constrained/blob/master/apps/simpleserver.c) sample app in the iotivity-constrained repo, it still can be discovered and accesses from the iot-rest-api-server running iotivity-node, even we didn't build the iotivity-constrained with SECURE=1

jimmy-huang, 2017-08-22:
@ttzeng As I pointed out, there are also discovery issues that are related to OCF 1.0 spec, and I guess you are using master on iotivity-contrained?  We are using a commit that's quite old and not OCF 1.0 compliant.  I think @nagineni is looking into that.

jimmy-huang, 2017-08-22:
Also, I think the reason we didn't update iotivity-contrained is that its master has been rebased off master of Zephyr and they've updated their API, were we are still targeting 1.8.0 release, so we have to stay with an older commit for API compatibility for 0.4 release, we'll probably fix this for the next release.

nagineni, 2017-08-22:
IoTivity 1.3.0 is compliance with the OCF 1.0 standards, so we should start using latest iotivity-constrained which has OCF 1.0 spec changes. I am working on updating the dependency in ZJS.

I also tried secure communication between iotivity-constrained simpleserver and iotivity simpleclint running both on Linux machine and noticed that it is failing. Kishen mentioned that he is working on updating IoTivity-Constrained’s security layer to OCF 1.0 spec.

xiaoyu3x, 2017-12-26:
@ttzeng 
This issue isn't reproduced and  'iot-rest-api-server' can discover OCF resource now,  please closed it.

closed 2018-01-01 by ttzeng