intel/zephyr.js #1483: [net] Can not create connection as tcp4 client on a101

cuiyanx, 2017-08-21:
### Description

I test tcp4 client on a101, but always can not connect to server.
 

### Test Code

Client on a101:
```
// Copyright (c) 2017, Intel Corporation.

// Run this test case and run simple TCP server on linux
//     TCP server: /tests/tools/test-tcp4-server.py
// Add soft router on linux:
//     ip -4 route add 192.0.2/24 dev eno1
// Add IPv4 address on linux:
//     ip -4 addr add 192.0.2.2/24 dev eno1

console.log("Test socket connection as TCP client for IPv4");

var net = require("net");
var assert = require("Assert.js");

var IPv4Address = "192.0.2.2";
var IPv4Port = 9876;
var addressOptions = {
    port: IPv4Port,
    host: IPv4Address,
    localAddress: "192.0.2.1",
    localPort: 8484,
    family: 4
}

var socket = new net.Socket();

assert(typeof socket === "object" && socket !== null,
       "SocketObject: be defined for TCP client");

socket.on("data", function(buf) {
    console.log("receive data: " + buf.toString("ascii"));
});

var SocketconnectFlag = true;
socket.on("connect", function() {
    if (SocketconnectFlag) {
        assert(true, "SocketEvent: be called as 'connect' event");

        SocketconnectFlag = false;
    }

    console.log("socket connection is connected");

    var Count = 0;
    var Timer = setInterval(function() {
        if (Count < 7) {
            socket.write(new Buffer("hello\r\n"));

            console.log("send data: 'hello'");
        }

        if (Count === 7) {
            socket.write(new Buffer("close\r\n"));

            console.log("send data: 'close'");

            assert.result();

            clearInterval(Timer);
        }

        Count = Count + 1;
    }, 3000);
});

socket.on("close", function() {
    console.log("socket connection is closed");
});

var socketonErrorFlag = true;
socket.on("error", function(err) {
    if (socketonErrorFlag) {
        assert(true, "SocketEvent: be called as 'error' event");

        socketonErrorFlag = false;
    }

    console.log("socket connection error: " + err.name);

    if (err.name === "NotFoundError") {
        setTimeout(function() {
            socket.connect(addressOptions, function() {
                assert(true, "SocketObject: connect server successful");
            });
        }, 5000);
    }
});

console.log("Add soft router on linux:");
console.log("    ip -4 route add 192.0.2/24 dev eno1");
console.log("Add IPv4 address on linux:");
console.log("    ip -4 addr add 192.0.2.2/24 dev eno1");

console.log("TCP client will be connecting after 30s on K64f board");
setTimeout(function() {
    socket.connect(addressOptions, function() {
        assert(true, "SocketObject: connect server successful");
    });
}, 30000);

```
Server on Linux:
```
# !usr/bin/python
# coding:utf-8

import time
import socket
import SocketServer

def main():
    host = "192.0.2.2"
    port = 9876
    print "Creating socket server on " + host + " port " + str(port) + "..."

    addr = (host, port)
    Timeout = 80
    bufSize = 1024

    mySocket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
    mySocket.settimeout(Timeout)
    mySocket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    mySocket.bind(addr)
    mySocket.listen(5)

    connection, clientaddr = mySocket.accept()
    print "Connected from: ", clientaddr

    while 1:
        try :
            Data = connection.recv(bufSize)
            Data = Data.strip()
            print "Got data: ", Data

            time.sleep(2)

            if Data == "close":
                connection.close()
                print "close socket"

                break
            else:
                connection.sendall(Data)
                print "Send data: ", Data
        except KeyboardInterrupt :
            print "exit server"
            break
        except :
            print "time out"
            continue

if __name__ == "__main__" :
    main()

```
 

### Steps to Reproduction

1. Build and flash a101 as client.
2. Add IPv4 address and route on Linux.
3. Run python code on Linux as server.
4. Reboot a101 and check console logs.
 

### Actual Result

Logs:
```
Test socket connection as TCP client for IPv4
PASS - SocketObject: be defined for TCP client
Add soft router on linux:
    ip -4 route add 192.0.2/24 dev eno1
Add IPv4 address on linux:
    ip -4 addr add 192.0.2.2/24 dev eno1
TCP client will be connecting after 30s on K64f board
PASS - SocketEvent: be called as 'error' event
socket connection error: Error
socket connection error: Error
socket connection error: Error
socket connection error: Error
socket connection error: Error

```
 

### Expected Result

Can connect to the server.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06cce05 | Arduino 101 | Aug 21, 2017 | Fail

 

### Additional Information

grgustaf, 2017-08-29:
The bug as stated is fixed by #1505, but a new bus fault crash occurs later on in the test on K64F.

grgustaf, 2017-08-30:
Also fixed that crash in another patch in #1505.

closed 2017-08-31 by grgustaf

qiaojingx, 2017-08-31:
Verified with commit a0b1217 on Arduino101 over Ethernet board. This issue is fixed.

cuiyanx, 2018-01-10:
Fixed and closed

PR #1505 (2017-08-31): [net] Fix socket_connect to support IPv4 connections
The function was returning an error immediately in the IPv4 case, if no
network context was set in the socket handle.

Fixes #1483, but then reveals another bus fault error on K64F.
commit 131d79b1 (2017-08-31): [net] Fix socket_connect to support IPv4 connections

The function was returning an error immediately in the IPv4 case, if no
network context was set in the socket handle.

Fixes #1483, but then reveals another bus fault error on K64F.