intel/zephyr.js #1488: [dhcp] dhcp does not work with ENC28J60

qiaojingx, 2017-08-23:
### Description

dhcp function is broken on Arduino101 with ENC28J60 Ethernet Board. 
 
### Test Code

[WebSocketServer4.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js);
[test-ws4-server-dhcp.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws4-server-dhcp.js);
[test-dhcp-ws-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-dhcp-ws-manual.js);
[TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js);
 [test-tcp4-server-DHCP.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server-DHCP.js);

### Steps to Reproduction

1. connect ENC28J60 to Arduino101 by wiring the below pins on boards.
```
CS  -> IO10
SI -> IO11
SO -> IO12
SCK -> IO13
VCC -> 3.3v
GND -> GND
INT -> IO4
```
2. Connect ENC28J60 Ethernet Board and PC to the same router.
3. Flash Arduino101 with test sample.
```
$ make JS=samples/TCPEchoServ4_DHCP.js FORCE=zjs_net_l2_enc28j60.json
$ make dfu
```
4. Connect TCPEchoServ4_DHCP.js using net-tools.
`$ nc <DHCP IP> 4242`
 
### Actual Result

[image]

### Expected Result

 dhcp works on Arduino101 with ENC28J60 Ethernet Board.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06cce05 | Arduino 101 | Aug 23, 2017 | Fail
 
### Additional Information

grgustaf, 2017-08-29:
Zephyr JIRA #1753 covered this issue and it was dropped as "Won't Do" because A101 is no longer a reference platform of concern. I git bisected the problem in Zephyr and it's a result of Zephyr commit db11fcd, net_pkt / net_buf changes. So if we care enough, we could look further into that.

grgustaf, 2017-08-29:
This was fixed by zephyrproject-rtos/zephyr#1284 which will presumably go into 1.9.

grgustaf, 2017-08-29:
Oh weird, apparently semaphores were broken back at that db11fcd commit; when I update to 1.8 w/ your patch, it works great!

haoxli, 2017-08-30:
@grgustaf So we will remove those tests in #1488 from 0.4 testing scope. Is that OK?

grgustaf, 2017-09-05:
@haoxli sounds good, yes.

PR #1504 (2017-09-13): [net] Allow DHCP to work on A101 w/ ENC28J60
A bug in Zephyr prevents this from working currently, but it has been
fixed in zephyrproject-rtos/zephyr#1284. This just enables DHCP for
platforms other than K64F. Also, a minor style/convention tweak.

Fixes #1488
commit e8079103 (2017-09-13): [net] Allow DHCP to work on A101 w/ ENC28J60 (#1504)

A bug in Zephyr prevents this from working currently, but it has been
fixed in zephyrproject-rtos/zephyr#1284. This just enables DHCP for
platforms other than K64F. Also, a minor style/convention tweak.

Fixes #1488
closed 2017-09-13 by grgustaf