intel/zephyr.js #1489: [ocf] The data type does not support arrays and decimal and string array

cuiyanx, 2017-08-23:
### Description

I set properties of resource as various types of data to test data transmission. I can not get the correct data that data type are arrays and decimal and string array. I always get "undefined".
 

### Test Code

Sample:
[/samples/thermostat/th02-ocf.js](https://github.com/01org/zephyr.js/blob/master/samples/thermostat/th02-ocf.js)

Simple test as server:
```
var ocf = require("ocf");
var server = ocf.server;

console.log("Test OCF server");

var testProperties = {
    num: 1024,
    negative: -1024,
    decimal: 80.1234,
    array: [512, 1024],
    boolean: true,
    string: "test",
    stringArray: ["test", "string", "array"]
}

var resourceInit = {
    resourcePath: "/test/data",
    resourceTypes: ["test.data"],
    interfaces: ["/oic/if/rw"],
    discoverable: true,
    observable: true,
    secure: true,
    slow: false,
    properties: testProperties
}

function onRetrieve(request, observe) {
    request.respond(testProperties);
}

server.on("retrieve", onRetrieve);

server.register(resourceInit).then(function(resource) {
    console.log("Registered resource test data");
}).catch(function(error) {
});

ocf.start();

```
 
Simple test as client:
```
var ocf = require('ocf');
var client = ocf.client;

ocf.start();

client.findResources({ resourceType:"test.data" }).then(function(resource) {
    console.log("findResources 'test.data' was successful, deviceId=" + resource.deviceId);

    client.retrieve(resource.deviceId, {observable: false}).then(function(res) {
        console.log("Resource path:" + res.resourcePath);
        console.log("Resource data:");
        console.log("    num: " + res.properties.num);
        console.log("    negative: " + res.properties.negative);
        console.log("    decimal: " + res.properties.decimal);
        console.log("    array: " + res.properties.array);
        console.log("    boolean: " + res.properties.boolean);
        console.log("    string: " + res.properties.string);
        console.log("    stringArray: " + res.properties.stringArray);
    }, function(error) {
    });
}, function(error) {
    console.log("findResources() returned an error: " + error.name);
});

```

Log:
```
findResources 'test.data' was successful, deviceId=7b9a6109-cb9a-4109-5b9b-61096b9b6109
Resource path:/test/data
Resource data:
    num: 1024
    negative: -1024
    decimal: undefined
    array: undefined
    boolean: true
    string: test
    stringArray: undefined

```
### Steps to Reproduction

 

### Actual Result

Get "undefined".
 

### Expected Result

Get the correct data.
 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 06cce05 | Arduino 101 | Aug 23, 2017 | Fail

 

### Additional Information

closed 2017-08-25 by grgustaf

qiaojingx, 2017-08-28:
Verified with commit 6c160b5d7. This issue is fixed.
[image]

cuiyanx, 2018-01-10:
Fixed and closed

PR #1497 (2017-08-25): [ocf] Fixed OCF encoding issues
Added support for decimal values and arrays for property values in
OCF responses, this fixes a issue where different array types will
have wrong value when encoded or missing proper encoding.  Also
fixed the client to properly re-create the JS object from decoded
reponses.

Fixes #1489
commit 237b367a (2017-08-25): [ocf] Fixed OCF encoding issues (#1497)

Added support for decimal values and arrays for property values in
OCF responses, this fixes a issue where different array types will
have wrong value when encoded or missing proper encoding.  Also
fixed the client to properly re-create the JS object from decoded
reponses.

Fixes #1489