intel/zephyr.js #1491: [tcp] bus fault error when running test-tcp6-client.js on K64F

qiaojingx, 2017-08-23:
### Description

This is a regression issue. BUS FAULT error occurs when running test-tcp6-client.js on K64F.

### Test Code

[test-tcp6-client.js ](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js);

### Steps to Reproduction

1.  `make BOARD=frdm_k64f JS=tests/test-tcp6-client.js`
2. `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`
3.  Add soft router and IPv6 address on linux:
```
ip -6 route add 2001:db8::/64 dev eno1
ip -6 addr add 2001:db8::2/64 dev eno1
```
4. `python tests/tools/test-tcp6-server.py`

### Actual Result

BUS FAULT error occurs after connection close
[image]
d.
 
### Expected Result

No BUS FAULT error. 

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 2439525 | FRDM_K64F | Aug 23, 2017 | Pass
master | 40f4274 - 06cce05 | FRDM_K64F | Aug 23, 2017 | Fail

### Additional Information

grgustaf, 2017-08-29:
Reproduced this and unfortunately it's not fixed with my latest patches.

grgustaf, 2017-08-30:
Fixed in #1505

closed 2017-08-31 by grgustaf

qiaojingx, 2017-08-31:
Verified with commit a0b1217 on FRDM-K64F. This issue is fixed.

commit a0b1217a (2017-08-31): [net] Check for NULL server_h pointer in client socket

In a recent patch that restructured connection tracking for server
connection sockets, I forgot that client sockets use the same code, so
for these cases server_h is NULL. Also, still needed to track the list
of client connections in a global for the timeout feature to work as
implemented.

Also, merge post_closed into release_close since it was the only
caller. And add some comments for future clarity.

Fixes #1491, as well as the bus fault revealed after #1483 was fixed