intel/zephyr.js #1498: [timer] Running test-timers.js on ArduinoDUE and F411RE cause BUS FAULT

qiaojingx, 2017-08-25:
### Description

When running [test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js) on ArduinoDUE and F411RE, BUS FAULT occurs, a lot of garbled log is printed continuously.

### Test Code

[test-timers.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)
 
### Steps to Reproduction

1.  F411RE:
```
$ make JS=tests/test-timers.js BOARD=nucleo_f411re
$ cp outdir/nucleo_f411re/zephyr.bin /media/<username>/NODE_F411RE/
```

2. ArduinoDUE:
```
$ make BOARD=arduino_due JS=tests/test-timers.js
$ bossac -p ttyACM0 -e -w -v -b outdir/arduino_due/zephyr.bin
```

### Actual Result

 F411RE:
[image]

ArduinoDUE:
[image]

### Expected Result

No error occurs.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 528a6f0 | Arduino Due | Aug 25, 2017 | Fail
master | 528a6f0 | Nucleo F411RE | Aug 25, 2017 | Fail
 
### Additional Information

grgustaf, 2017-08-29:
We have Due boards and should get F411RE soon, so we can take a look at this. But these haven't been supported board to date so won't fix for 0.4.

brianjjones, 2017-09-05:
Nucleo F411RE works fine for me on the latest master, we might have already fixed the issue.  I see:

PASS - clearInterval: intervalID does not exist
UPASS - clearTimeout: timeoutID does not exist
UPASS - setInterval: set delay
PASS - clearInterval: intervalID
PASS - setInterval: set delay and optional arg
PASS - setTimeout: set delay
PASS - setTimeout: set delay and optional arg
PASS - setTimeout: set 20 arg for timeout
PASS - clearTimeout: timeoutID
TOTAL: 9 of 9 passed

@qiaojingx  are you still seeing this issue?

qiaojingx, 2017-09-06:
Verified with latest commit 087938a in master. This issue is not reproduced on Nucleo F411RE and Arduino due anymore.
Close

closed 2017-09-06 by qiaojingx