intel/zephyr.js #1502: [ashell] Redundant message outputs

xiaoyu3x, 2017-08-28:
## Issue description
Redundant message is printed on terminal and it affect the next input.

## Test Code
[ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)

## Steps to reproduce
1. `make ROM=256 ashell; make dfu`
2. Flash ashell into 101 board, reset it and open port:
    `$ screen /dev/ttyACM0 115200`

## Actual Result
Redundant message is printed.
[image]

## Expected Result
No redundant message outputs.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [6c160b5](https://github.com/01org/zephyr.js/commit/6c160b5d77c32b89e0fd165a644b4cd243d01929) | Arduino 101 | Aug 28, 2017 | Fail

brianjjones, 2017-08-30:
I'm betting its the same thing that causes it to be printed for issue #1458

brianjjones, 2017-08-30:
This will be fixed by #1510

qiaojingx, 2017-09-01:
Verified with commit a0b1217 +#1510， this issue is not reproduced.

xiaoyu3x, 2017-09-20:
This issue will be closed again when PR #1510 is merged.

brianjjones, 2017-10-13:
Looks like #1576 will be the new fix.

xiaoyu3x, 2017-10-17:
Verified with commit [be3ebf7](https://github.com/01org/zephyr.js/pull/1576/commits/be3ebf7c43843a99eb7229d7c3700628de666291)+[dd371e7](https://github.com/01org/zephyr.js/pull/1576/commits/dd371e794c65b3ec61100a61493d29f3c5d512f0) +[#1576](https://github.com/01org/zephyr.js/pull/1576)， this issue is not reproduced.

PR #1576 (2017-10-19): [ble] Fixed ble in Ashell
This fixes the issue where bluetooth doesn't work in Ashell,
since when running in Ashell, bt_enable() cannot be called until
the Bluetooth module is initialized.  This also fixes the error message
showing up when IDE is launched.

Fixes #1502
closed 2017-10-19 by grgustaf