intel/zephyr.js #1507: [tcp/websocket/udp] Default ble address is missing in tcp/websocket/udp server.

qiaojingx, 2017-08-30:
### Description

This is a regression issue. When running tcp6 server, websocket6 server and udp6 server on A101. Bluetooth adapter cannot search for ble address of this server.
 
### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js);
[UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js);
[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js);
[test-udp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server.js);
[test-udp6-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server-manual.js);
[test-ws6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws6-server.js)
 
### Steps to Reproduction

1. `make JS=samples/TCPEchoServ6.js ROM=256 `
2. `make dfu`
3. `hcitool lescan` to search ble address of this server.

### Actual Result

Bluetooth adapter cannot search for ble address of this server.
[image]

### Expected Result

Should get server address like:
[image]

 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bf70e36 - 6c160b5 | Arduino 101 | Aug 30, 2017 | Fail
master | d3f0434 | Arduino 101 | Aug 30, 2017 | Pass
### Additional Information

If add the following code in these samples, this issue will not reproduce.
```
var net_cfg = require('net-config');
net_cfg.setBleAddress('FF:EE:DD:CC:BB:AA');
```
Is that by design that default ble address be removed? if so, please update sample to set ble address.

brianjjones, 2017-09-01:
#1514 should fix this.  The  default address was actually being set, the problem was ipss_init() and ipss_advertise() shouldn't have been getting called in zjs_net_config_default.  Those are now called in the main.c file.  The reason it worked when you added the net-config code is that those calls get commented out when BUILD_MODULE_NET_CONFIG is defined.

qiaojingx, 2017-09-01:
Verified with commit a0b1217+ #1514. This issue is not reproduced.

closed 2017-09-01 by grgustaf

LWY931119, 2017-09-05:
When I run TCPEchoServ6.js on my arduino101. The device name is not 'ZJS Device',it is 'Zephyr',why?

haoxli, 2017-09-05:
@LWY931119 Yes, you're right, it should be 'Zephyr' for `TCPEchoServ6.js`. Because of the issue, we cannot get the expected screenshot with device name 'Zephyr', so use OCF's screenshot instead. Sorry for your confusion, we will update the expected result.

commit d04ae005 (2017-09-01): [net] Stop initializing ipss twice (#1514)

The default address was actually being set, but there
were other methods being called that have been moved
to the main file, and needed to be removed otherwise it caused
havoc.

Fixes #1507