intel/zephyr.js #1508: Variable BLE_ADDR is not working if no default ble address

qiaojingx, 2017-08-30:
### Description

When building tcp6 server, websocket6 server and udp6 server with BLE_ADDR flag. Ble address isn't added. Bluetooth adapter cannot search for ble address of this server, too.
 
### Test Code

[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js);
[UDPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/UDPEchoServ6.js);
[WebSocketServer.js](https://github.com/01org/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js);
[test-udp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server.js);
[test-udp6-server-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-udp6-server-manual.js);
[test-ws6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ws6-server.js)
 
### Steps to Reproduction

1. `make JS=samples/TCPEchoServ6.js ROM=256 BLE_ADDR="FF:E1:D2:C3:B4:A5"`
2. `make dfu`
3. `hcitool lescan` to search ble address of this server.

### Actual Result

Bluetooth adapter cannot search for ble address of this server, too.

### Expected Result

Should get server address like: `FF:E1:D2:C3:B4:A5 ZJS Device`

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | bf70e36 - 6c160b5 | Arduino 101 | Aug 30, 2017 | Fail

### Additional Information

If add the following code in these samples to add default ble address. Then build with `BLE_ADDR` flag, it will work.
```
var net_cfg = require('net-config');
net_cfg.setBleAddress('FF:EE:DD:CC:BB:AA');
```

brianjjones, 2017-09-05:
This should be fixed by #1507, I'll test it today and see if that's the case.

brianjjones, 2017-09-05:
This works for me now with the #1507 fix.  Please test it again @qiaojingx when you get a chance, thanks!

qiaojingx, 2017-09-06:
Verified with commit 087938a, this issue is not reproduced. 
Close

closed 2017-09-06 by qiaojingx