intel/zephyr.js #1512: [tcp] Cannot receive TCP packet when send single data pack by net-tools

qiaojingx, 2017-08-31:
### Description

 When sending single data pack by command `echo "hello" | nc 192.0.2.1 4242` to [TCPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4.js), tcpserver cannot recevie tcp packet(`(ERROR) Cannot receive TCP packet (family 10), ret=-9`). By if connect with `nc 192.0.2.1 4242` first, then send data, this issue will not reproduce. The same issue also occurs when running tcp server6. But It works fine on Arduino101.

### Test Code

[TCPEchoServ4.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4.js);
[TCPEchoServ4_DHCP.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ4_DHCP.js);
[TCPEchoServ6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPEchoServ6.js)
 
### Steps to Reproduction

1.  `make JS=samples/TCPEchoServ4.js BOARD=frdm_k64f`
2.  `cp outdir/frdm_k64f/zephyr.bin /media/<username>/MBED/`
3.  add router and address on Linux
`ip route add 192.0.2.0 dev <ethX>`
`ip addr add 192.0.2.2 dev <ethX>`
4. send single data pack by net-tools: 
`echo "hello" | nc 192.0.2.1 4242`

### Actual Result

 TCP4:
[image]

TCP6:
[image]

### Expected Result

Arduino101:
[image]
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a0b1217 | FRDM-K64F | Aug 31, 2017 | Fail
master | a0b1217 | Arduino101 | Aug 31, 2017 | Pass

### Additional Information

LWY931119, 2017-09-05:
When I run TCPEchoServ4.js on my arduino101 I can't  connect to it by 'nc 192.0.2.1 4242',but TCPEchoServ6.js works fine .Whether arduino101 support TCPEchoServ4.js?
I already added router and address on Linux  by ' ip route add 192.0.2.0 dev  bt0 'and ' ip addr add 192.0.2.2 dev bt0'.

qiaojingx, 2017-09-05:
@LWY931119, TCPEchoServ4.js is only supported on arduino101 over Ethernet board, not over ble. Because 6lowpan is used for arduino over ble, and it is IPv6.  For IPv4 testing, you need a ENC28J60 Ethernet Board. Please refer to [ENC28J60 Ethernet Board](https://github.com/01org/zephyr.js#enc28j60-ethernet-board) for IPv4 testing on Arduino101.

LWY931119, 2017-09-05:
@qiaojingx Thank you for your help. I still want to know there are any Http sample can run on Arduino101. The zephyr/samples/net/http_server says 'This sample is meant to be compilable in QEMU only'.

qiaojingx, 2017-09-05:
http api is a further feature for ZJS. Feature #374.
For zephyr OS, http should supported in either QEMU or real device like Freedom Board (FRDM-K64F) according to https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/http_server.

qiaojingx, 2017-09-07:
Verified with latest commit 087938a in master, this issue occurs on Arduino101 over Ethernet(not reproduce with d04ae00). Data "hello" is not received.
[image]

grgustaf, 2017-09-07:
Fixed by #1522

qiaojingx, 2017-09-08:
Verified with commit 79e8164 + #1522. `Cannot receive TCP packet` issue is fixed. But new issues are brought.
An minor error is printed on K64F:
[image]

Cannot connect by net-tools second time on Arduino101 over Ethernet.
[image]

grgustaf, 2017-09-08:
There's nothing wrong with the first one - maybe we need to think about whether this should be an "error" or not. The socket has already been closed by the remote end by the time the socket write is attempted, so the write fails. But there's no getting around that.

The second one I'll have to look into I guess.

grgustaf, 2017-09-08:
The problem w/ A101 + Eth is fixed in #1523

closed 2017-09-11 by grgustaf

qiaojingx, 2017-09-13:
Verified with commit 2a71022. This issue is not reproduced.

commit 74a79d84 (2017-09-11): [net] Mark socket as "closing" between EOF and close event

Then check for this closing flag to know that write will fail and not
bother attempting it.

Fixes #1512 (the follow-on report of failure on A101 w/ Ethernet)