intel/zephyr.js #1515: [ocf] CPU exception has a high probability of occurrence when running multiple sensors.

xiaoyu3x, 2017-09-01:
## Issue description
CPU exception happens 9/10 times when I running multiple sensors.

## Test Code
[multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js)

## Steps to reproduce
1. Build sample multiple.js and flash it to Arduino 101.
2. Reset Arduino 101 and open port:
    `$ minicom -D /dev/ttyUSB0`

## Actual Result
'CPU exception 13' appears .
[image]

## Expected Result
There isn't error message and multiple sensors work well.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [e3e4dde](https://github.com/01org/zephyr.js/commit/e3e4dde59c0595447107b391ede8f23c517b0ad3) | Arduino 101 | Aug 14, 2017 | Pass 
master | [a0b1217](https://github.com/01org/zephyr.js/commit/a0b1217a0d8128c255bbab6b6a9aaa062df7ba9c) | Arduino 101 | Sep 1, 2017 | Fail

xiaoyu3x, 2017-09-01:
Multiple sensors work normally when flashing zjs-server/multiserver2.js.
There are 3 sensors in multiserver2.js, but there are 6 sensors in multiserver.js.

jimmy-huang, 2017-09-28:
Is this still reproducible? I just tried master cbe7e8f98910f07a9d30bea7fccf880b449bb3d3, and I couldn't reproduce this.

xiaoyu3x, 2017-09-29:
hi, @jimmy-huang 
I retest 10 times and there are only 3 times to occur this issue with commit [cbe7e8f](https://github.com/01org/zephyr.js/commit/cbe7e8f98910f07a9d30bea7fccf880b449bb3d3) and multiserver.js.
I am glad to this low repetition rate.

jimmy-huang, 2017-09-29:
Ok, I'll keep this a P2 as 6 resources are definitely stretching the limit on this Arduino 101, so anything we add could starting breaking it here or there.

jimmy-huang, 2018-01-25:
@xiaoyu3x is this still reproducible?

xiaoyu3x, 2018-01-25:
@jimmy-huang 
This issue isn't produced with PR #1789 when run make ROM=272 JS=multiserver.js.

xiaoyu3x, 2018-01-25:
This issue isn't reproduced, so closed it

closed 2018-01-25 by xiaoyu3x