intel/zephyr.js #1518: [tcp] tcpclient cannot connect to server on A101 and qemu

qiaojingx, 2017-09-04:
### Description

This is a regression issue. When running tcpclient on A101 and qemu, it can't connect to tcpserver at most of time. 

### Test Code

[TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js);
[test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js);
 
### Steps to Reproduction

 1. Build [TCPClient6.js](https://github.com/01org/zephyr.js/blob/master/samples/TCPClient6.js) on A101 over BLE or Ethernet,
 2. add router on Linux
3.  patch [tests/tools/tests-tcp6-server.py](https://github.com/01org/zephyr.js/blob/master/tests/tools/test-tcp6-server.py) to use port 4242 which TCPClient6.js is expected
4. execute `python tests/tools/tests-tcp6-server.py` in 10s
### Actual Result

Cannot connect to server at most of time.
[image]

Sometimes, [test-tcp6-client.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-client.js) can connect, but data transfer not work well.
[image]

### Expected Result

Connection can be established. Data transfer works well
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | fef1e19 - d04ae00 | Arduino 101 | Sep 4, 2017 | Fail
master | fef1e19 - d04ae00 | QEMU | Sep 4, 2017 | Fail
master | 8a3853f | Arduino 101 | Sep 4, 2017 | Pass
master | 8a3853f | QEMU | Sep 4, 2017 | Pass 
### Additional Information

qiaojingx, 2017-09-07:
Verified with latest commit 087938a in master, TCPClient6 cannot connect to Server on Arduino101 over BLE.
It is able to connect on Arduino101 over Ethernet, but get `cannot acquire send_buf` error when send data.

grgustaf, 2017-09-07:
Hmm, seems to work fine for me on A101 with Ethernet at least w/ latest. Possibly retry with PR #1522. I guess I'll try BLE also.

grgustaf, 2017-09-07:
Shoot after I ran "make pristine" and rebuilt again, it no longer works. I do see the send_buf error.

grgustaf, 2017-09-07:
Actually, that seems to be a fluke. Now it has worked several more times, including after another "make pristine".

qiaojingx, 2018-01-25:
Verified with commit 0c0af9b in master. TCPClient6 on Arduino101 and 96b_carbon over BLE still cannot connect to TCPServer6.

qiaojingx, 2018-02-05:
Retest with commit 4cf09c5 in zjs-0.5 branch. TCPClient6 on Arduino101 over BLE cannot connect to TCPServer6. But it works on Arduino101 over Ethernet. Will it be fixed in zjs-0.5 branch?