intel/zephyr.js #1525: close opcode not handled in websockets

grgustaf, 2017-09-11:
If you test w/ tests/test-ws6-server.js on K64F (run node tests/tools/test-ws6-client.js on Linux), after it's done and the close occurs, you'll see a garbage string printed in the last payload on the device. (Meaning you built with VARIANT=debug to dump packets.)

Also it complains about unknown opcode 0x08. We should handle that close opcode and treat the payload as the int that it is instead of a string.

cuiyanx, 2017-09-13:
This is regression issue, look #1234 .

brianjjones, 2017-09-22:
This should be fixed by #1526.  I can't currently test it myself because my frdm_k64f won't let me flash it.

grgustaf, 2017-12-01:
@cuiyanx can you verify?

cuiyanx, 2017-12-04:
@grgustaf 
Adding debug flag, throw out error.
No debug flag, `ReceiveClose:59395`

cuiyanx, 2018-01-03:
@grgustaf @brianjjones 
I test again with #1739 PR, can build with `VARIANT=debug`. But `code = buffer data` in [/src/zjs_web_sockets.c](https://github.com/intel/zephyr.js/blob/master/src/zjs_web_sockets.c) line 478, is that right? I think that `code = opcode = 0x08`. Please check again.