intel/zephyr.js #1530: [tcp] Out of memory when running test-tcp[4/6]-server.js in qemu

qiaojingx, 2017-09-13:
### Description

 When running test-tcp4-server.js and  test-tcp6-server.js in qemu, crash happens after print 27 results.

### Test Code

[test-tcp6-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp6-server.js);
[test-tcp4-server.js](https://github.com/01org/zephyr.js/blob/master/tests/test-tcp4-server.js);
 
### Steps to Reproduction

1. start net-tools on Linux
   terminal1: `./loop-socat.sh`
   terminal2:`sudo ./loop-slip-tap.sh`
2. `make BOARD=qemu_x86 qemu JS=test-tcp6-server.js`
 
### Actual Result

Crash happens: `qemu: fatal: Trying to execute code outside RAM or ROM at 0xaaaaaaaa`
[image]
 
### Expected Result

 No crash happens. All the results can be printed.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | Comment
---|---|---|---|---|---
master | 49f13c3 - 2a71022 | Qemu | Sep 13, 2017 | Fail | out of memory
master | 2cfcb66 | Qemu | Sep 13, 2017 | Pass | #1519 
 
### Additional Information

grgustaf, 2017-09-13:
Interesting, the 0xaaaaaaaa address it shows is, I believe, the value they write to initialize of the stack. So it would seem to mean we're jumping to an address from an uninitialized value from the stack.

grgustaf, 2017-09-13:
Since it's just QEMU I don't think I'll hold up 0.4 for this.

grgustaf, 2017-09-13:
This is probably a dup of #1519, something's corrupted and so it can appear in various ways. (Haven't tried to debug it yet, just hypothesizing.)

qiaojingx, 2017-10-11:
Verified with commit 0a17d94, this issue is not reproduced.
Closed

closed 2017-10-11 by qiaojingx