intel/zephyr.js #1545: [ide] Multil-evel catalog causes ROM overflow

qiaojingx, 2017-09-20:
### Description

If `zephyr.js` is placed under multil-evel catalog, `make ide ROM=256` will cause ROM overflow. 
 
### Test Code

[ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md)
 
### Steps to Reproduction

1.  Create folder with multil-evel catalog like: `/home/neusoft/aaa/bbb/ccc/ddd/...`
2.  Try to build `ide` in `zephyr.js` which is placed under these catalogs.

### Actual Result

 If the path is `/home/neusoft/aaa/bbb/ccc/ddd/zephyr.js`, `make ide ROM=256` will get error:
[image]

 If the path is `/home/neusoft/aaa/bbb/ccc/ddd/eee/zephyr.js`, `make ide ROM=256` will get error:
[image]

### Expected Result

The path of zephyr.js doesn't affect ide building.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 174f35f | Arduino 101 | Sep, 20, 17 | Fail
zjs-0.4 | b11dc55 | Arduino 101 | Sep, 20, 17 | Fail
 
### Additional Information

Test with commit 151a2ae. `make ide ROM=256` will get RAM overflow and `make ide ROM=255` will get ROM overflow if `zephyr.js` located under `/home/neusoft/aaa/bbb/.../ooo/ppp/`.

brianjjones, 2017-11-15:
@grgustaf I'm pretty sure this a build issue, any chance you know off hand what the problem is?  This isn't related directly to IDE other than the fact that it pulls in all the modules, using up more memory.

grgustaf, 2017-12-01:
This is because we use __file__ macro so the big long path of each file gets stored in the binary. @nagineni fixed it I think, by disabling that __file__ usage except for debug builds.

@qiaojingx can you check the latest?

qiaojingx, 2017-12-04:
Verified with commit cbec79d in master, this issue is fixed. But it still exists in zjs-0.4 branch. 
Will the zjs-0.4 branch be fixed, too?

grgustaf, 2018-01-24:
No, let's not worry about it for 0.4 at this point. We'll be releasing 0.5 soon.

closed 2018-01-24 by grgustaf