intel/zephyr.js #1548: [ocf] Expected resource cannot be found with iot-rest-api-server@v0.6.0

xiaoyu3x, 2017-09-21:
## Description

Expected resource cannot be found with iot-rest-api-server@0.6.0 and node@6.11.0

## Steps to Reproduce

1.Build sample [led.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/led.js) and flash it to Arduino 101.
2.Connect to BLE, install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server)
3.Run `./oic-get /res`, after getting all ocf resources.
4.Check "/a/led" whether is found. 

## Actual Result

"/a/led" resource can't be found.
The following is information by port($run: `minicom -D /dev/ttyUSB0`) 
[image]

## Expected Result

"/a/led" resource can be found.
No error on the port.

## Test Build

Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [292e6bb](https://github.com/01org/zephyr.js/commit/292e6bb9294a01131aaa9fceeb7e27f98473e462) | Arduino 101 | Sep 21, 2017 | Fail

nagineni, 2017-09-21:
/oic/res response size might be too large to get accommodated in the TX/RX buffers and causing the discovery failure. Please try to do the resource discovery with specific "rt" option and see if it works.

$ `./oic-get /res?rt=oic.r.switch.binary`

xiaoyu3x, 2017-09-22:
@nagineni 
The resource can be discovered with command `$ ./oic-get /res?rt=oic.r.switch.binary`
Return value:
`[{"di":"68fa6309-b9fa-4309-48fb-630957fb6309","links":[{"href":"/a/led","rt":["oic.r.switch.binary"],"if":["oic.if.baseline"],"p":{"bm":1,"secure":false}}]}]`

But there are some commands that can not be used.
```
$ ./oic-get /a/led?di=***
$ ./oic-post /a/led?di=*** true.json/false.json
```
Even through add `rt=**` to behind them:  
```
 $ ./oic-get /a/led?di=*** rt=**
$ ./oic-post /a/led?di=*** true.json/false.json rt=**
```
return result also is:
```
Resource retrieve failed: findResources: unexpected response
HTTP: 404
or
[ ]
```
How to use these commands to realize my target?
The following is return result, hope to have some help to you
[image]

nagineni, 2017-09-22:
> The resource can be discovered with command $ ./oic-get /res?rt=oic.r.switch.binary

Nice.

> $ ./oic-post /a/led?di=*** true.json/false.json

I can see in the logs that POST is also working fine.

> $ ./oic-get /a/led?di=***

GET is still failing, and the reason is explained here: https://github.com/01org/zephyr.js/issues/1457#issuecomment-324599754 

We need to remove discovery `(/res)` on GET in iotitivy-node bindings so that the response will fit into the available buffers.

nagineni, 2017-09-22:
iotivity-node issue to remove discovery on GET: https://github.com/otcshare/iotivity-node/issues/154

jimmy-huang, 2017-09-29:
@nagineni I am having problems with the ?rt= using iot-rest-api-server@0.6.0, I always get:
localhost:8000 /res?rt=oic.r.switch.binary
[]
HTTP: 200

I tried other samples with different resource types and it's the same, I have not played with OCF since the 1.3.0 update, not sure what I am missing, do you have any idea?

iot-rest-api-server e789b4c7eeeaacbab83b597b9200716f93aa0e35
zephyr.js cbe7e8f98910f07a9d30bea7fccf880b449bb3d3
iotivity-constrained 0a85a8d7d0403700f0a64f8c7a175c1af742cfd0

wanghongjuan, 2017-10-10:
@nagineni Glad to see issue otcshare/iotivity-node#154 was fixed now, I tried with iotivity-node commit[  b839be4](https://github.com/otcshare/iotivity-node/commit/c3ecd6e9b50bd4139d9178636df94525199ba24a) and iot-rest-api-server commit [d77d94b](https://github.com/01org/iot-rest-api-server/commit/d77d94bbdf70d9e042ccbe866b3c5ece099041ce), the command:
```
$ ./oic-get /a/led?di=***
```
can work fine now, but for multiple sensors, the result is unstable, specially when the sensor number is more than two, the command:
```
$ ./oic-get /res?rt=oic.r.switch.binary
```
only can discover resource 2 in 10 times.
When one resource is found via this command, the serial port should output content including incoming and outgong information:
[image]
if no resource is found, I found there only has incoming information in the serial port output and got the same result with @jimmy-huang:
```
localhost:8000 /res?rt=oic.r.switch.binary
[]
HTTP: 200
```
It seems that the server has received the rest api request, but return an empty object without any error message, actually the resource is existed.

nagineni, 2017-10-10:
@wanghongjuan Thanks for testing. Looks like the response size is too large to get accommodated in iotivity-constrained’s buffers when you are running with the multiple sensors on A101.

wanghongjuan, 2017-10-10:
Do you have any solution can fix this case? There is one issue #1515 about mutilple sensors, in this scenario, 3 sensors are OK, but 6 sensors are limited for A101

xiaoyu3x, 2017-10-12:
By command: `$ ./oic-get /a/led?di=***`  has worked, so closed it.
For multiple sensors issue, it will be tracked with bug #1566

closed 2017-10-12 by xiaoyu3x