intel/zephyr.js #1549: [ashell] Sometimes memory malloc fails

xiaoyu3x, 2017-09-21:
## Description

There are 3/10 times that memory malloc fails when running the IDE examples.

## Steps to Reproduce

1.Build IDE and flash it to Arduino 101.
2.Launch IDE, run any two program code in the examples and swith them randomly.

## Actual Result

```
acm> ashell/jerry-code.c:154 read_file_alloc():
(ERROR) malloc failed
```

## Expected Result

No "malloc failed" occurs.

## Test Build

Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [292e6bb](https://github.com/01org/zephyr.js/commit/292e6bb9294a01131aaa9fceeb7e27f98473e462) | Arduino 101 | Sep 21, 2017 | Fail

grgustaf, 2017-12-01:
This probably occurs because of a memory leak; @xiaoyu3x can you clarify whether this happens only when you repeat execution of scripts. Does it ever occur the first time you run an example after rebooting the device?

xiaoyu3x, 2017-12-04:
hi, @grgustaf 
I just retested 10 times with commit [cbec79d](https://github.com/01org/zephyr.js/commit/cbec79db327b51501806851685eb896739a9a424), but it wasn't reproduced. I think it is random and  probability is very low now.

grgustaf, 2018-01-24:
I'm closing since it wasn't reproduced; let's reopen if the problem comes back.

closed 2018-01-24 by grgustaf

xiaoyu3x, 2018-01-30:
This issue isn't reproduced with commit [d09e4b3](https://github.com/intel/zephyr.js/commit/d09e4b36d02f78c144b78b9b9194ad0e7305a632)