intel/zephyr.js #1550: Failed to build IDE image on zjs-0.4

nagineni, 2017-09-21:
`make ide ROM=256` is failing to build IDE image on zjs-0.4 branch. 

Error:
```
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/real-ld: zephyr.elf section `initlevel' will not fit in region `ROM'
/opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/libexec/i586-zephyr-elfiamcu/gcc/i586-zephyr-elfiamcu/6.2.0/real-ld: region `ROM' overflowed by 140 bytes
collect2: error: ld returned 1 exit status
```

Note: If I increase the ROM size, the ARC image doesn't fit in available space.

nagineni, 2017-09-27:
Looks like this is reproducible only if the sources are located in a deep directory path (e.g /home/sudarsana/web/zephyrproject/zephyr.js).

grgustaf, 2017-10-05:
Good catch.

PR #1570 (2017-10-06): [utils] Do not include __FILE__ macro in non debug builds
Using __FILE__ macro in non debug code is increasing the
image size depending on where the sources are built, so
this patch removes that macro from the non debug code.

Fixes #1550
commit 7198395a (2017-10-06): [common] Do not include __FILE__ macro in non debug builds (#1570)

Using __FILE__ macro in non debug code is increasing the
image size depending on where the sources are built, so
this patch removes that macro from the non debug code.

Fixes #1550
closed 2017-10-06 by grgustaf