intel/zephyr.js #1552: [ocf] Expected resource cannot be found with iot-rest-api-server@v0.5.0

xiaoyu3x, 2017-09-22:
## Description

Expected resource cannot be found with iot-rest-api-server@0.5.0 and node@6.11.0

## Steps to Reproduce

1.Build sample [led.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/led.js) and flash it to Arduino 101.
2.Connect to BLE, install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server)
3.Run `./oic-get /res`, after getting all ocf resources.
4.Check "/a/led" whether is found. 

## Actual Result

"/a/led" resource can't be found.

## Expected Result

"/a/led" resource can be found.

## Test Build

Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [cbe7e8f](https://github.com/01org/zephyr.js/commit/cbe7e8f98910f07a9d30bea7fccf880b449bb3d3) | Arduino 101 | Sep 22, 2017 | Fail 

## Comment
Device doesn't receive requestion through device's port.
[image]

nagineni, 2017-09-27:
IoTivity-Constrained in master is updated to latest and it is OCF 1.0 standards compliance, so you should be testing against the iot-rest-api-server version 0.6.0 which is based on IoTivity 1.3.0.

xiaoyu3x, 2017-09-28:
According to nagineni advice, so closed it.

closed 2017-09-28 by xiaoyu3x