intel/zephyr.js #1553: [ashell] 'grove_lcd' module can't be found

xiaoyu3x, 2017-09-25:
## Issue description
'grove_lcd' module can't be found  when running [GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js) in ashell. 

## Test Code
[GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js)

## Steps to reproduce
1. Connect grove lcd and Arduino 101 board.
2. Flash ashell into 101 board, open minicom port. 
3. Copy [GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js) to ashell and run it.
4. When running this sample on Web IDE, get the same result as ashell

## Actual Result

**Ashell and Web IDE**
`Not SupportedError: module not found:'grove_lcd'`
[image]

## Expected Result
Grove LCD works normally and no error.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [cbe7e8f](https://github.com/01org/zephyr.js/commit/cbe7e8f98910f07a9d30bea7fccf880b449bb3d3) | Arduino 101 | Sep 25, 2017 | Fail 
master | [174f35f](https://github.com/01org/zephyr.js/commit/174f35f2a022037ca6559e5f66c961022210ac82) | Arduino 101 | Sep 25, 2017 | Pass

jimmy-huang, 2017-09-28:
We took grove lcd out of ashell/IDE recently due to growing ROM size with ashell, that it will no longer fit, so yeah, grove_lcd is not expected to work out of the box, so this is not a bug.

xiaoyu3x, 2017-09-29:
OK, according to developer's advice, this bug will be closed.

closed 2017-09-29 by xiaoyu3x