intel/zephyr.js #1555: ringbuf put errors when running SensorBLEDemo.js for a long time

qiaojingx, 2017-09-25:
### Description

When running [SensorBLEDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js) on A101 for 4hours, `ringbuf put errors` will happens.
 
### Test Code

[SensorBLEDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/SensorBLEDemo.js)

### Steps to Reproduction

1. `make JS=samples/SensorBLEDemo.js ROM=256`
2. `make dfu`
3. Connect it by nRF Connect app and Subscribe in it.

### Actual Result

[image]

### Expected Result

No error happens.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
zjs-0.4 | b11dc55 | Arduino101 | Sep 25, 2017 | Fail

### Additional Information

grgustaf, 2017-12-01:
In this case these errors do not really matter. It just means that out of four hours worth of sensor samples, 13 of them were lost. It looks like we're sending samples at 20Hz, which means 288k samples in 4 hours... so this is a failure rate of 0.004%.

It probably happens occasionally during garbage collection or something.

qiaojingx, 2018-02-06:
Verified with commit 4cf09c5 in zjs-0.5 branch. This issue is not reproduced after running 4hours.
Closed.

closed 2018-02-06 by qiaojingx