intel/zephyr.js #1556: [ocf] Segment fault when running OcfClient on Linux to find resource from OcfServer on K64F for long time

qiaojingx, 2017-09-25:
### Description

When running [OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js) on Linux to find resource from [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on K64F for 4hours, `Segment fault` will happens on Linux side. But re-run [OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js) again on Linux, connetion still can be setup.
 
### Test Code

[OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js);
[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);

### Steps to Reproduction

1. Build [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on K64F.
2. Build [OcfClient.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js) on Linux.
3. Connect K64F to PC with Ethernet cable.
4. `./outdir/linux/release/jslinux samples/OcfClient.js --noexit`

### Actual Result

After running almost 4hours, `Segment fault` came out.
[image]

### Expected Result

No error happens.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
zjs-0.4 | b11dc55 | Linux | Sep 25, 2017 | Fail

### Additional Information

jimmy-huang, 2018-01-18:
@qiaojingx can you re-test this with latest master and with the #1769 patch?

qiaojingx, 2018-01-18:
Verified with commit 4b23a47 + #1769. This issue is not reproduced.

closed 2018-01-22 by qiaojingx