intel/zephyr.js #1560: [ashell] The program is hanged after running 'BMI160Accelerometer.js' for a period of time

xiaoyu3x, 2017-09-26:
## Issue description
There is 9/10 times that 'BMI160Accelerometer.js' will be hanged  after running it for a period time  in ashell or IDE.

## Test Code
[BMI160Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Accelerometer.js)

## Steps to reproduce
1. Flash ashell into 101 board and start it : `minicom -D /dev/ttyACM0`
2. Copy 'BMI160Accelerometer.js' to ashell and run it.
4. When running this sample on Web IDE, get the same result as ashell

## Actual Result

The program is hanged after running 'BMI160Accelerometer.js' for a period of time

## Expected Result
'BMI160' works normally and no hang.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [cbe7e8f](https://github.com/01org/zephyr.js/commit/cbe7e8f98910f07a9d30bea7fccf880b449bb3d3) | Arduino 101 | Sep 26, 2017 | Fail

grgustaf, 2017-12-01:
@xiaoyu3x how long were these runs?

xiaoyu3x, 2017-12-04:
hi, @grgustaf 
I just retested 10 times with commit [cbec79d](https://github.com/01org/zephyr.js/commit/cbec79db327b51501806851685eb896739a9a424) and it costs about 5 min every time, but it wasn't reproduced.

xiaoyu3x, 2017-12-11:
This issue isn't reproduced on [1b970c8](https://github.com/01org/zephyr.js/commit/1b970c822e051d900777c0b66e3723e8c40cd74f), so closed it.

closed 2017-12-11 by xiaoyu3x