intel/zephyr.js #1561: [ashell] command boot not working properly

qiaojingx, 2017-09-26:
### Description

1. After set [BMI160Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/BMI160Accelerometer.js) as boot file in ashell, then reboot device, `ringbuf put errors` will print at the beginning. And sensor start delay.

2. Not all the samples can start automatically after be set by boot. e.g. [BLE.js](https://github.com/01org/zephyr.js/blob/master/samples/BLE.js)

### Test Code

[Ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md);
 
### Steps to Reproduction

1. `make ashell ROM=256`
2. `make dfu`
3. load the code of samples to ashell with name as test.js
4. `acm>boot test.js`
5. `acm>reboot`

### Actual Result

1. ringbuf put errors print first if boot BMI160Accelerometer.js in ashell
[image]

2. BLE service not start if boot BLE.js in ashell although can see `BLE sample...` in `/dev/ttyUSB0`.
[image]

### Expected Result

No error and boot file can start automatically after reboot.

### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | cbe7e8f | Arduino 101 | Sep 26, 2017 | Fail
master | b11dc55 | Arduino 101 | Sep 26, 2017 | Fail
 
### Additional Information

xiaoyu3x, 2017-10-10:
This issue is reproduced when using commit [3ca5e23](https://github.com/01org/zephyrjs-ide/commit/3ca5e23effd1ef9dddcfa49a963c660fd225cf0a)

brianjjones, 2017-10-26:
This is fixed by #1615

closed 2017-10-31 by grgustaf

qiaojingx, 2017-11-01:
Verified with commit e19ed4f. This issue is fixed.

commit e19ed4f6 (2017-10-31): [ashell] Fix for ashell boot (#1615)

Currently ashell boot doesn't work with javascript with callbacks
or timers.  This is because once the boot.cfg JS gets called, it
goes into an infinite loop waiting for uart. Until it gets the
connection, it won't return.  This means the main loop doesn't get
control back, meaning callbacks are not processed.  This breaks
the code up a bit so that it checks if the connection is ready
when uart_process gets called, if not try to connect.  If the
connection fails, just return and go through the main loop again.
Continue until a connection is made.

Fixes #1 [... cut]