intel/zephyr.js #1566: Resource can only be retrieved at the first time when running multiple sensors together

xiaoyu3x, 2017-09-28:
## Description

Resource  can only be  retrieved at the first time with iot-rest-api-server@0.6.0 and node@6.11.0 when running multiple sensors together，after resource can not be found again.

## Steps to Reproduce

1.Build sample [multiserver2.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver2.js) and flash it to Arduino 101.
2.Connect to BLE, install and launch [iot-rest-api-server](https://github.com/01org/iot-rest-api-server)
3.Run `./oic-get /res?rt=oic.r.temperature`  to get temperature resource.
4. After temperature is found, execute `./oic-get /res?rt=oic.r.temperature`  to get temperature again.

## Actual Result

No resources can be found anymore except the first time.
The following are commands output on terminal.
 
[image]

The following are port( run `minicom -D /dev/ttyUSB0`) return.
[image]

## Expected Result

Resource can always be found.

## Test Build

Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [cbe7e8f](https://github.com/01org/zephyr.js/commit/cbe7e8f98910f07a9d30bea7fccf880b449bb3d3) | Arduino 101 | Sep 28, 2017 | Fail

xiaoyu3x, 2017-09-29:
Sometimes, resource can't be retrieved at all, this occurs especially if flash [multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js) into 101 board.

jimmy-huang, 2018-01-08:
This is a ram issue with a101 with BLE and OCF enabled, it's only stable to run 1 or 2 sensors registered, anything with 3 or more sensors will be unstable.

jimmy-huang, 2018-01-11:
This is related to RAM issue on the Arduino101, is this reproducible on the 96b_carbon?

xiaoyu3x, 2018-01-12:
hello, @jimmy-huang 
I can't build OCF relative image successfully with 96_carbon.
Build command: `make BOARD=96b_carbon JS=**ocf**.js`
Result:
[image]

Does 96b_carbon support OCF?
Is it  a issue about 96b_carbon?

jimmy-huang, 2018-01-23:
@xiaoyu3x Is this reproducible on k64f?  We can't test on 96b_carbon because we don't have the tool to flash the bluetooth firmware over that board.

jimmy-huang, 2018-01-23:
@xiaoyu3x Can you retest this with #1785, and using make JS=multiserver2.js ROM=260?  multiserver2.js doesn't require ARC image, so you can bump up the size even further to get around the size problem.

xiaoyu3x, 2018-01-24:
@jimmy-huang
This issue isn't reproduced when using `make ROM=256 JS=multiserver2.js` with #1785  on A101.
 K64f doesn't support PWM, so can not use it to verify this issue.

xiaoyu3x, 2018-01-25:
This issue isn't reproduced, so closed it.

closed 2018-01-25 by xiaoyu3x