intel/zephyr.js #1579: [Mac] Fail to build Linux and Qemu target on Mac

qiaojingx, 2017-10-11:
### Description

This is an regression issue and it is blocked weekly test on MacOS. Building linux and qemu target on Mac failed. 

### Test Code

All samples and tests

### Steps to Reproduction

1. Building linux on MAC by command: `make BOARD=linux`
2. Building qemu on MAC by follow instructions: [building-qemu-and-arduino-101-targets](https://github.com/01org/zephyr.js#building-qemu-and-arduino-101-targets)

### Actual Result

1. Building linux target failed on Mac.
[image]

2. Building qemu target failed on Mac.
[image]

 ### Expected Result

Building successfully.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a36b2c9 - cd3f44e | qemu & linux | Oct 11, 2017 | Fail
master | 4b77505 | qemu & linux | Oct 11, 2017 | Pass

### Additional Information

This issue is caused by commit a36b2c9.

jimmy-huang, 2017-10-13:
This is borked by a36b2c90502351f4f8a5e930249ce63c0e0be549, since the it uses the librt timer library which doesn't exists on Mac, only in Linux, @jprestwo

jimmy-huang, 2017-10-24:
Reverted commit with this patch #1608

qiaojingx, 2017-10-26:
Verified with commit ce4036c + #1608. Make linux target succeed, but make qemu target still failed.
Qemu:
[image]
[image]

jimmy-huang, 2017-10-26:
@qiaojingx I updated patch, but am having problems installing qemu on my mac, can you see if it fixes the build error?

qiaojingx, 2017-10-27:
Verified with commit cac4c3a + updated #1608 . Unforthnately, make qemu target still failed.

[image]

jimmy-huang, 2017-10-27:
@qiaojingx updated patch

qiaojingx, 2017-10-30:
Verified with commit 6d751ea + updated #1608 , warnings on Mac are gone, but make qemu target still fails because of `linker error`.
[image]

closed 2017-11-01 by grgustaf

jimmy-huang, 2017-11-02:
@qiaojingx can you open up a separate issue for qemu?  Btw, did you try with the latest instructions on setting up crosstool-ng with 1.9?  http://zephyr-docs.s3-website-us-east-1.amazonaws.com/online/1.9.0/getting_started/installation_mac.html?  I was struggling to get crosstool-ng to build that's why I couldn't test qemu.

qiaojingx, 2017-11-03:
New issue for qemu is #1631.  I also run into trouble when executing `ct-ng build` by following the latest instructions on setting up cross tool-ng with 1.9.  Issue #1631 is on old cross tool-ng.

PR #1608 (2017-11-01): [timers] Revert patch #1542 for linux timers
The new timer patch #1542 uses librt which doesn't build on MacOS,
this patch revert linux builds to previous old timer
implementation for compatibility and adds a FIXME for all the places
the patch was reverted.  We need to find a way to add a cross-platform
implementation for timers that will work on both Linux and MacOS.

Fixes #1579
commit 6149191a (2017-11-01): [timers] Revert patch #1542 for linux timers (#1608)

The new timer patch #1542 uses librt which doesn't build on MacOS,
this patch revert linux builds to previous old timer
implementation for compatibility and adds a FIXME for all the places
the patch was reverted.  We need to find a way to add a cross-platform
implementation for timers that will work on both Linux and MacOS.

Fixes #1579