intel/zephyr.js #1582: ringbuf put errors when running I2CBMP280.js and HeartRateDemo.js on A101

qiaojingx, 2017-10-11:
### Description

This is an regression issue and probably caused by #1542.` ringbuf put errors` prints when running [HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js) and [I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js) on A101.

### Test Code

[HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js);
[I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js)
 
### Steps to Reproduction

1. `make JS=samples/I2CBMP280.js`
2. `make dfu`

### Actual Result

[image]

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | a36b2c9 - cd3f44e | Arduino101 | Oct 11, 2017 | Fail
master | 4b77505 | Arduino101 | Oct 11, 2017 | Pass

### Additional Information

brianjjones, 2017-10-25:
Looks like the setInterval isn't finishing before it gets called again.  I'll look to make sure there isn't a problem with callbacks.  It may also be that 10 is too low a timer number.

brianjjones, 2017-10-25:
Looks like callbacks are fine, the demo timer is just going faster than the callbacks can be handled - hence the error message.  #1614 will fix this.

qiaojingx, 2017-10-27:
Verified with commit cac4c3a + #1614. Error in I2CBMP280.js has been fixed. Could you change interval time in HeartRateDemo.js, too?

grgustaf, 2017-10-27:
This seems to indicate we have a performance regression because we used to do this work in 10ms apparently and now it takes 200ms? Rather than mask the problem (:D) let's look into why. @brianjjones, could you try testing this just before and after #1542 to see if that's what caused the problem.

grgustaf, 2017-12-01:
@brianjjones ping on this one, just note I don't think we're done here.

jimmy-huang, 2018-01-12:
@cuiyanx is this still reproducible?

cuiyanx, 2018-01-15:
@jimmy-huang 
Yes, this issue can be reproduced on both cases. And [/samples/Stepper.js](https://github.com/intel/zephyr.js/blob/master/samples/Stepper.js) case looks like this issue.
Log:
```
Stepper Motor sample...

554:(ERROR) 1 ringbuf put errors (last rval=-122)

554:(ERROR) 15 ringbuf put errors (last rval=-122)

554:(ERROR) 47 ringbuf put errors (last rval=-122)

554:(ERROR) 110 ringbuf put errors (last rval=-122)

```