intel/zephyr.js #1586: [ocf] Ocf Server cause CPU exception after zephyr update to 1.9

qiaojingx, 2017-10-12:
### Description

After update zephyr to 1.9.0, `CPU exception` occurs when running Ocf Server on A101.
 
### Test Code

[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js);
[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js);
[OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js);
[test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js)
 
### Steps to Reproduction

1. `make JS=samples/OcfServer.js ROM=256`
2. `make dfu`
### Actual Result

1.[OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js):
[image]

 2.[OcfSensorServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfSensorServer.js) and [OcfMultiSensor.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfMultiSensor.js):

No console log is printed.

3.[test-ocfserver.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ocfserver.js):
[image]

### Expected Result

OcfServer started without error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | ac89871 | Arduino 101 | Oct 12, 2017 | Fail
master | 0a17d94 | Arduino 101 | Oct 12, 2017 | Pass
### Additional Information

xiaoyu3x, 2017-10-13:
This issue is reproduced with commit [1d5a99d](https://github.com/01org/zephyr.js/commit/1d5a99d5e2b8b09602e79cd4c1cd1c260eaf224b)

xiaoyu3x, 2017-10-13:
There isn't any output information on minicom when flashing [temperature.js]( https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/temperature.js),[multiserver.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver.js) and [multiserver2.js](https://github.com/01org/SmartHome-Demo/blob/master/ocf-servers/zjs-servers/multiserver2.js) to Ard 101 with [1d5a99d](https://github.com/01org/zephyr.js/commit/1d5a99d5e2b8b09602e79cd4c1cd1c260eaf224b)

closed 2017-11-15 by grgustaf

qiaojingx, 2017-11-16:
Verified with commit dcf56a9. OcfServer works on Arduino101, but it still doesn't work on FRDM-K64F. Create a separate issue(#1660) to track it.

commit da045395 (2017-11-15): [ocf] Updated OCF 6lowpan (#1620)

Updated iotivity-constrained with latest 6lowpan fixes,
and updated 6lowpan configs to work with latest Zephyr 1.9

Fixes #1586, #1565, #1481, #1237